fesch/CanZE #198: Double BT connection ...

fesch, 2015-11-08:
Sometimes, when starting CanZE, a first BT connection is made immediately (and successfully), but a second one continuously tries to connect too, so it interferes with the first one and no command can be send to the remote BT device.

yoh-there, 2015-11-08:
I am always suspicious when it IMMEDIATELY states it is connected. Seems to happen when I hit return too often, then immediately restart CanZe, as if there is "stuff left".

fesch, 2015-11-08:
Thanks for the hint ...

fesch, 2015-11-09:
Bug confirmed by Carsten (via mail)

yoh-there, 2015-11-09:
This seems a "channel" problem within the existing pairing. It could be either the dongle or Android being confused about a still open channel.

Current thinking is Android still "thinks" there is a channel open and CanZE opening a new one to the dongle fails.

There is a slight chance though that it is the dongle having this problem. I am copy-pasting some email text here to document our thinking, just in case we need it later on.

In my experience when things go haywire, my first action is to switch off the dongle for 5 seconds or so. If it is the dongle, there is absolutely no way to correct that, as ANY command to reset it would be lost in transit anyway. The only option left might be to switch OFF Android's BT adapter for 5 seconds (not the channel, the entire BT adapter). That might convince the "HC-05"-ish module in the KW902 to tear down all it's channels.

PieterZoe, 2015-11-09:
I experience the same problem: when starting CanZe the bluetooth connection gets established quickly. Then after a minute or so, CanZe no longer updates the screens (frozen data) while it still shows " Connected". I then switch off/on the dongle which usually helps.
This problem does not always happen...

fesch, 2015-11-09:
On my way home, CanZE got stuck twice:
- The first time I tried to switch off the dongle, which *did not* help. I needed to restart CanZE.
- The second time I immediately restarted CanZE, which helped.

yoh-there, 2015-11-09:
OK, so it is not the dongle. To be honest, I am relieved at that.

fesch, 2015-11-09:
Today's test were horrible. The connection broke every few minutes and it did *not* recover.

I now hopefully got it sorted out now, but this thread debugging is really PITA!! You always have to think in which thread you are right now, who is calling who or starting who and am I not joining myself (which, of course, resides in a full stop of that thread ...)

yoh-there, 2015-11-09:
Wow, had a quick look at the Diff's, that was an awful lot of work.

yoh-there, 2015-11-10:
I am sorry to say today was not good. A CanZE thread crashes @ startup when BT is not enabled, and when things go wrong, they go wrong in a bad way. Had to restart both CanZE and the dongle (it once got stuck with the OBD led fixed on. Dang.

fesch, 2015-11-10:
I confirm ... :-(

fesch, 2015-11-10:
It seams it is ELM related (meaning the ELM327.class, not the dongle). I will implement some "debug logging" to save, session based, all emitted messages into a file, so I could traceback later what went wrong. 

I can't have connected a computer while driving, so I think this is the only way to get into it ...

yoh-there, 2015-11-10:
Please drop a message here once you've pushed that. I can try to obtain some data too if that helps.

fesch, 2015-11-10:
Push with debugLogger is in ...

fesch, 2015-11-11:
This morning, CanZE did not crash nor did it hang. Everything worked seamlessly, but I'm not yet trusting the apparent stability!

yoh-there, 2015-11-11:
Fingers crossed! No car this morning, but I will try this afternoon.

fesch, 2015-11-11:
On my drive home today, I had a few timeouts, but CanZE recovered within seconds!

Spot them on the bottom graph?

[image]

yoh-there, 2015-11-12:
Short drive today. Very stable.

fesch, 2015-11-12:
:-)

fesch, 2015-11-12:
No further problem today. Even recovered between two trip, just shutting down the ELM and powering of my tablet. 

So I close this issue ...

closed 2015-11-12 by fesch

yoh-there, 2015-11-12:
Congrats man, I know that was NOT easy.