fesch/CanZE #218: Night mode

yoh-there, 2015-11-15:
Suggestion by @avataranedotas : Some devices do not have a light sensor. A nice feature would be an (easy to access) night mode. I have no idea how (hard that is) to implement.

ps: the S3 I run CanZE on does not have that problem. The light sensor is pretty good, and personally I use an app called "Night Filter" when things are still too bright. Actually, the toggle is sitting in my always accessible bottom menu bar.

In case we decide NOT to implement a night mode, we might add that as advice to the website.

fesch, 2015-11-16:
We even could read Zoe's light sensors ... but need to figure out how to control the brightness of the screen or whether it is possible to set some "global" screen filter without the need to recode all existing graphical stuff.

yoh-there, 2015-11-16:
Googled around a bit. Going directly without any helpers seems to be a real PIA, but it seems people have had good results with his

https://gist.github.com/slightfoot/c508cdc8828a478572e0

I confess I don't really understand exactly what is implied with the resource file edits, but it seems to be a start.

yoh-there, 2015-11-16:
Light sensors: Hmmmm, my simple Zoe "live" version, I don't think it has one.

fesch, 2015-11-16:
0x5EE
LeftSolarLevelInfo
RightSolarLevelInfo

yoh-there, 2015-11-16:
Very nice. I will check if the info there makes some sense in my car. What model do you have @fesch ?

fesch, 2015-11-16:
Beuhh ... the "blue" one :-D ... I will need to check, never looked that up. The car is the first one to be delivered to a customer here in Luxembourg ... 2013 ... must be the first model I think...

yoh-there, 2015-11-16:
To recognize this table is the easiest:
Life = basic, card needs to be put in the slot in the center console.
Zen = Life + Park assist distance control, card does not have to be shoved in the slot.
Intens = Zen + Park assist camera

yoh-there, 2015-11-16:
Our sources for frame 5EE do not match at all. I need to research this frame again top to bottom :-( I can't imagine I made such weird mistakes, but maybe I did. I suggest to put this part of the request on hold until I have done that. Of course I did not mean putting on hold how to display a night mode, just the automatic triggering.

fesch, 2015-11-16:
Ahhh ....

Then it's the Zen I have, because I do not need to insert the card
to the console to do anything with the car ;-)

Bob

fesch, 2015-11-16:
I got the information from the _other_ list ...

yoh-there, 2015-11-16:
Yes, and I trust that one more. But still, need to verify. And then we need to cross verify the models. I think the Zen has auto-headlights. The Life doesn't, so maybe the sensors are simply not there. I will investigate. This week, promise.

fesch, 2015-11-16:
I did a little research on the net: there seams nothing to be which can easily be added to an app to invert the colors with single line of code.

Android itself has an accessibility option to invert the colors, but I did not find on how to access this programmatically.

Thus the only way would be to do it manually. For the widgets, this will be quite easy, as I could replace or enclose each call to a color with a method that checks the actual mode and returns the corresponding color (either the color itself of it opposite). But for any other button or text, this method will be much more difficult.

yoh-there, 2015-11-16:
https://gist.github.com/slightfoot/c508cdc8828a478572e0 was not applicable?

I have no problem dropping the issue. There are tons of apps for night mode, enforcing a lower backlight and reduced color table. Sort of inverted flashlight. I use

https://play.google.com/store/apps/details?id=com.digipom.nightfilter

fesch, 2015-11-16:
Nope ... that's just a helper class that saves the given state or toggles it.

I will continue to dig a bit, but with "invert colors" I did not get that far ...

avataranedotas, 2015-11-16:
If there is no information from the light sensors just use the headlights stalk information, when the user turns on the headlights night mode is engaged.
Or you can check the phone time and see it it's after sunset.

yoh-there, 2015-11-16:
Smart approach. Though headlights would not work in he Nordic countries (light always on, by law). Still, it could be a setting: none - use headlight - use clock. The sunset clock formula is quite complicated but doable. Not that I am being lazy, but I think the light sensor approach will not work in the Zoe Life. I will figure it though.

It seems that implementing the actual dimmed screens is harder.

solmoller, 2015-11-16:
I am a bitt puzzled here: apart from detecting light (could be done in
numerous ways) the colour scheme is presently with a white background
everywhere. Changing that to a grey background would enable us to keep all
other display parameters unchanged.

Henrik

avataranedotas, 2015-11-23:
Bit 40 of free frame 5EE on the Fluence gives day/night mode, tested and confirmed.

For cars without sensor, if you can get the last know GPS position, you can use one of the many libraries for calculating sunrise/sunset time.
If you can't get the last know GPS position you could simply default to night after 8pm and before 6am.

yoh-there, 2015-11-23:
5ee.40 added to the spreadsheet.

Will check on my el cheapo Zoe model. If I don't forget :-)

GPS is overdone I would humbly suggest, if anything, a country should be enough, no minute-precision is needed.

fesch, 2015-11-23:
http://developer.android.com/training/location/index.html

yoh-there, 2015-12-06:
As I am plowing through all our frames, there is a frame representing the light sensors, and of course the light status. Funny, there is also a combined bit: it is on when there low light OR the driver switches on the lights.

Anyway. My proposition is to drop this issue (Sorry Alex): the triggering is either so so (for the cars without sensors it could only be done through the lights; my R-Link does that and I hate it), or we need to geolocate (well hello!!! For screen dimming!!!). And then the dimming itself seems far from trivial. My take on this is we are chasing a non-problem with way too much effort. Installing a ready made night filter such as I posted does the job quite nicely.

But that is truly all my not-so-humble opinion, feel free to disagree.

yoh-there, 2015-12-08:
Added this to the Q&A

"Q:My tablet does not have ambient light sensors. It's binding at night?
A:Most phones and tablets adjust the brightness of the display automatically. If it doesn't or you are not happy with it, our stance is that it affects all apps, and not just CanZE. We will not make a night mode specifically for CanZE. We would advice you to use a night mode app. Some are reporting good results with this one."

Closing.

closed 2015-12-08 by yoh-there