fesch/CanZE #219: Newer versions not working

fesch, 2015-11-16:
As reported here: http://renault-zoe.forumpro.fr/t4455p90-obd-ii-canbus#135387
Newer versions of CanZE do not work with certain dongles, but older versions do!

yoh-there, 2015-11-16:
I am not denying this is true, but since 11 October? We should have heard MUCH more complaints. I bet Ke2 has a non-compliant dongle. So does Chameleon. I apologize for being sceptical, but a lot of these people seem to be willing to spend days (and tons of forum posts) on "it doesn't work", where buying KV902/Maxiscan would 99% sure solve the problem. So, that would be my first advice.

Something else: we really need to press them to use gitbug issue tracker instead of responding in fora to real issues. I know I respond too (but not in this thread, no way, you know why), but it is a bad habit and all of us, and them, loose track.

yoh-there, 2015-11-16:
Reading the forum, I think the gentleman needs to pair his dongle. Instructions in the Q&A...........

fesch, 2015-11-16:
After a lot of mails with Carsten I come to the conclusion that there seams to be a problem. Older version work with his two dongles, but the newer do not!

yoh-there, 2015-11-16:
From the French forum, I got the impression that was a BT linking issue. Is this different? If BT link is NOT the problem, how about a Test Dongle screen print?

fesch, 2015-11-16:
Yes, it is different. Got a video. He has the same dongle I have (but different color - maybe the black ones are not good ;-))

I'm waiting for a screenshot as well as for a debug log file ...

fesch, 2015-11-16:
On his phone, he even does not see the three-dotted menu :-(

fesch, 2015-11-17:
For the french guy: he did not chose the right BT device!!

yoh-there, 2015-11-17:
:-/ So, now we have kaimaik's problem left open If there is anything I can do, let me know.

fesch, 2015-11-17:
Here are the two log files ...

[2015-11-17-17-37-33.txt](https://github.com/fesch/CanZE/files/37037/2015-11-17-17-37-33.txt)
[2015-11-17-17-37-11.txt](https://github.com/fesch/CanZE/files/37036/2015-11-17-17-37-11.txt)

fesch, 2015-11-17:
As far as I can see, on each request something goes wrong and the request is then empty.

I think I need to get that dongle for further debugging ...

yoh-there, 2015-11-17:
maybe it is a good plan to also log Device.lastInitProblem. I suspect a faulty processor in the dongle with an operational, paired BT module.

Last time that happened to me, the two wires of the crystal were shorted out. Symptom: NO double blink on startup.

yoh-there, 2015-11-17:
See http://canze.fisch.lu/faulty-konnwei-maxiscan-dongle/

fesch, 2015-11-17:
Yes, that's what I did: added some more debug messages and send him a new APK. We will see tomorrow.
OK for your tutorial, but this does not explain why older versions of CanZE work ... :-?

yoh-there, 2015-11-17:
Oh the old versions STILL work??????? I thought that was a myth, sorry. No, then forget about the broken dongle tutorial (though I will link to it in the Q&A)

fesch, 2015-11-17:
new log files ...

[2015-11-17-20-57-21.txt](https://github.com/fesch/CanZE/files/37147/2015-11-17-20-57-21.txt)
[2015-11-17-20-58-09.txt](https://github.com/fesch/CanZE/files/37148/2015-11-17-20-58-09.txt)

fesch, 2015-11-17:
He confirmed me that the dongle still works with an older version.

From the new logs, I read that the dongle does not return any data because it is being hit by a timeout.

Remote debugging is not that easy ...

yoh-there, 2015-11-17:
Well I did optimize the ATMA's quite agressively, but not the ISO-TP's. Any difference in that maybe?

yoh-there, 2015-11-17:
I am asking because frame repeat speed MIGHT be different in the Kangoo for several frame types? If anything, just (for him) increase the timeout you see in the get free frame routine :-/

fesch, 2015-11-22:
F*** ... why the hell does the line:
    frequencyMultiplicator+=0.1;
not give a warning when frequencyMultiplicator is being declared as "int" ???

yoh-there, 2015-11-22:
Because Java is so smart it recognizes is not a frequency but the inverse
:D :D :D

Sorry, couldn't resist.

yoh-there, 2015-11-22:
Teaser 8 published.

Let me know if/when you want to publish this weeks version, as I might want
to commit that blasted Kangoo temp screen on experimental, but it needs a
fix in fields init.

Question: Is it possible on the website to add the tapatalk plugin? I can't
and I don't know if it is at all possible through wordpress hosting. If it
is possible, could you have a stab at it please?

Jeroen

fesch, 2015-11-22:
Hi

I posted the new version some minutes ago, just before the teaser ...
So if you have new stuff to commit for the next release, just go ahead.

I will check for the tapatalk pluging ...

Bob

fesch, 2015-11-23:
Carsten's dongle (the same as mine, just in black ...) works fine on my Zoé, so this problem is Kangoo related! I think that's the direction we need to dig further.

*joke mode ON*
Is here anyone that could send me a Kangoo ZE for testing?
*joke mode OFF*

OK, I need to get someone over here who has one. Maybe I will ask the local Renault dealer. I'm not afraid to the a "no" ....

yoh-there, 2015-11-23:
Hehe!
So, it is timing. And the conclusion should be that for certain frames, the timing in the fluence/kangoo is different than the Zoe.

Here is what I propose to do (two options really).
1. Simply check the half-october timeouts and revert, or
2. Have someone check the repeat intervals of the frames. It would be fair to compile a list with the free frames we now use to keep the workload in check (I think it is 10 frames max).

And then we need to extend the Frames tab & class with a "repeat_interval_flukan" column

yoh-there, 2015-11-23:
Alexandre promised to do some timing measurements

fesch, 2015-11-24:
As far as I can read from my latest log files, the dynamic increasing timeout does not improve the situation, meaning that there are not less problems that occur. (Just once more to underline it: CanZE recovers from these problem immediately!)

yoh-there, 2015-11-24:
So, is the conclusion (for now) that it is NOT a timeout (or time-in) problem?

fesch, 2015-11-24:
Not on the Zoé ...

I will really need to check on a Kangoo ...

yoh-there, 2015-11-24:
OK, so, just to be sure.......
1. One problem is specific for the Kangoo (probably Fluence too, same systems). **IF** that is timing, it is a;ready "solved" with the new Frames (re)initialization, we just need the number. Alexandre already promised to provide some.
2. An entirely other problem is the lockups. I have them now VERY often. Seems like everything we do makes thing worse.

Is my understanding (two separate problems, and one (number 1) probably being solved soon) correct?

fesch, 2015-11-24:
For (1): today I've send back Carsten's dongle, so as soon as he has it, I will ask him to test the new version with your modifications, so we will see.

My addition with the dynamically increasing timeout seams no to hurt, so I will leave it there an do some more tests.

What type of lookups do you have? I've got none these last days ...

yoh-there, 2015-11-24:
Usualy "silence". No flashing LED's, no responses in the debug field. I still have no idea where the hang occurs. It almost seems like the query queue is empty. I have not much time for testing at the moment unfortunately, but I guess I need to add a ton of debugging (logging) statements.

fesch, 2015-11-24:
In a particular screen?

yoh-there, 2015-11-24:
Hmmmm. I don't think so. If so, it must be in the driving screen as I usually have that up. Another indication is maybe that sometimes thing become very unresponsive (many seconds before fields update) and when I look at the dongle, there is usually not much activity then. That is also why I suspect a queuing issue, but really, I cannot be sure.

If it would be because of an overloaded input (dongle blaring data), the new "time-in" code should prevent that, and it doesn't.

fesch, 2015-11-26:
I modified my Arduino Due to measure the interval between frames (in the red marked column). They match pretty well what we actually have for the Zoé. So, next thing to do is to get access for some seconds to the CAN-bus of a Kangoo ;-)
[image]

yoh-there, 2015-11-26:
Way cool! I have modified the Frames code already to take a new set for the non-Zoe Z.E. cars.

yoh-there, 2015-11-26:
Modified code #233 to get a better grip and not let the intervalMultiplicator out of hand.

fesch, 2015-11-26:
I asked someone living in my town and owning a Kangoo EV to get access to it for some minutes. Waiting for his response ... if negative, I will try to get a talk with the manager of the local Renault dealer ...

yoh-there, 2015-11-28:
I am sensing a pattern. When I get in my car in the morning and start driving, I get a hangup pretty fast. Usually within the first couple of minutes. Then I restart both the dongle and CanZE and then usually it runs fine. I would imagine R-Link's Bluetooth would be interfering, but I switched that off and it didn't help. Could it have to do with me always switching off Bluetooth on my phone in the evening so it is actually CanZE firing it up? I will experiment with that hypothesis a bit.

fesch, 2015-11-29:
I got access granted to a Kangoo, so I now just have to fix some appointment ;-)

andyfras, 2015-12-04:
I haven't checked for a couple of weeks but there seems to be a problem with my devices on the current build. 

My Moto G phone returns either zero or NaN in every field.

My Asus TF201 reads some values, but no firmware (something about ecu labels)
The available range in miles shows about 10 decimal places and shows a different number each time I try it (e.g. 59, 23, 14).

This is on a ZOE with a Konnwei adapter. The adapter is correctly paired and the host light is flashing while Canze is running.

yoh-there, 2015-12-04:
That sounds pretty serious. Could you provide us with some extra data please? What would be really helpful would be:
- screenshots of the offending screens and the settings screen;
- log files of a session going bad. To make one, go to the settings screen and tick "Write debug log to SD card. The logs are usually made in \phone\Canze (from memory)

Thanks!

andyfras, 2015-12-04:
[image]
[image]
[image]
[data2015-12-04-14-52-30.txt](https://github.com/fesch/CanZE/files/52164/data2015-12-04-14-52-30.txt)

Strangely, it seems to be working better on the TF201 now. I have some firmware data, and the available mileage is more consistent (to 14 decimal places), but you can see from the graph that it does an unexplained dip at one point.

Similarly, it has started working on the Moto G phone too, showing the same data as the TF201.

I have no idea what was upsetting it before. I had tried rebooting, replugging the adapter, and even reloading the app. I'm sorry for causing you any consternation.

yoh-there, 2015-12-04:
No worries and thanks for the update. A "hickup" is not always preventable though. The range (decimals) is probably a result from converting to miles, which I haven't really tested. I will have a look into that and consider it fixed in the next release.

fesch, 2015-12-04:
I'm worried by the upper graph (SoC): the dots should be spaced up by half a minute, so they are really to close to each other :-/

Could you perhaps do two test:
#1:
- start CanZE
- go immediately to the "consumption" screen and stay for at least 5 minutes
- post a screen shot of that screen ;-)

#2:
- start CanZE
- go to some random screens (switch a few time)
- go to the "consumption" screen and stay for at least 5 minutes
- post a screen shot of that screen ;-)

yoh-there, 2015-12-04:
Range formatting fixed #245 

@fesch : I bet that has to do with the email I send you 2 days ago

fesch, 2015-12-04:
Yes ... I will have a look at this this evening. Had a really stuffed up week and not as much time as I would have liked to ;-)

andyfras, 2015-12-04:
[image]
[image]
[image]
I've also included the Battery screen, where you can see that the cell voltage is not displayed, as the display has clipped at 4V. Perhaps 2V could be used as the X axis?

andyfras, 2015-12-04:
[image]
Here is my Firmware page for reference. There seems to be a lot missing. My ZOE is October 2013 in case that has any relevance.

fesch, 2015-12-04:
@yoh-there: While recoding the scheduler, I forgot to clear a list in clearFields() :-(

yoh-there, 2015-12-06:
@andyfras would you be so kind to recheck when you have today's version in? I think, after 53 comments, we are ready to close this issue ;-)

andyfras, 2015-12-07:
I won't be able to do it until tomorrow, as I have no time today.

kaimaik, 2015-12-07:
Hab heute die neue Version im Kangoo probiert.
Es kommen wieder Daten :-)

andyfras, 2015-12-08:
I've run the tests again. The first thing I noticed was the typo on the main page where it now says Bell Voltages instead of Cell voltages. Then I wondered where Firmware had gone, but found it under Experimental.

There are some strange glitches on the Consumption graphs.

[image]
[image]
[image]

fesch, 2015-12-08:
Regarding the consumption graph, this behavior can be reproduced like this:
switch of the car, but leave the dongle active and CanZE too

Explanation: After a while all CAN connected computers go to sleep and no more data is being send over the bus. When woken up, the first few frames seam to contain invalid data, of some init-code. I'm not yet sure about this ...

yoh-there, 2015-12-09:
Closing. Most issues are solved, and the thread has become too unspecific.

closed 2015-12-09 by yoh-there