fesch/CanZE #221: Website (marked as being) inscure??

yoh-there, 2015-11-19:
A friend said http://canze.fisch.lu was being marked as insecure/infected/whatever by F-secure. Anyone able to confirm that? More detail would be welcome.

solmoller, 2015-11-21:
Correct, F-Secure does see a problem:

https://search.f-secure.com/search?query=http%3A%2F%2Fcanze.fisch.lu%2F

Virustotal does not:
https://www.virustotal.com/da/url/fc44b4ca5bc4f4763dd4572d453976473ecd0e5107a50564e1f54e3eddadb9cf/analysis/1448137042/

I have contacted F-Secure and made them aware of the apparently false
positive, let's see what happens

Henrik

solmoller, 2015-11-24:
The problem persists

Let's try pushing, please go to the f-secure link and report the site as
wrongly categorised

Henrik

yoh-there, 2015-11-24:
Will do.

Unless there is something else we're overlooking, posting the apk there might be a trigger too. If possible, I suggest to remove it and host the old version either on github, or, if that is logically impossible (because of sequencing) I (or google ;-) ) can provide hosting which is is irrelevant to any F-secure silliness.

yoh-there, 2015-11-24:
By the way, it would be nice if the "Report a search result that was rated incorrectly" would actually take you to a page that would allow that. It just shows a page where you can report suspicious files or url's. If I overlooked something, please report here, as I would gladly report, and would like to know why they rate us like this. F**ers.

fesch, 2015-11-24:
I've removed the CanZE APK's, but left Alexander's in place.

yoh-there, 2015-11-24:
@avataranedotas Can we host "your" APK's elsewhere? (see this issue for the rationale)

fesch, 2015-11-24:
false positive reported. Let's see what they will answer ...

fesch, 2015-11-24:
Hello,

Thank you for your submission.

The submitted website has been verified to be clean and the appropriate rating is now updated.

The updated rating will take effect in the next database update cycle.We apologize for any inconveniences caused and we will be closing this ticket now.

Should you have further questions, please feel free to contact us again.

Best Regards,
Kiran Krishnappa
Junior Malware Analyst
F-Secure Security Labs

Visit our Labs blog at https://labsblog.f-secure.com/
Give and get advice in our F-Secure Community at https://community.f-secure.com
Contact Support at https://www.f-secure.com/support

avataranedotas, 2015-11-24:
Do I still need to move my apk's ?

yoh-there, 2015-11-24:
I would think not as they said they would remove the rating?

fesch, 2015-11-25:
Is now marked as "safe" --> closing

fesch, 2015-11-25:
@avataranedotas: No, you can leave it there ...

closed 2015-11-25 by fesch