fesch/CanZE #229: Endless loop

fesch, 2015-11-24:
Frames.init code depends on Fields.init code which depends on Frames.init code ... --> out of memory

yoh-there, 2015-11-24:
Oops

yoh-there, 2015-11-24:
It is of that bloody getCar() that doesn't belong in Fields anyway ;-)

What do you propose? Move to global in Main, or a Car Class? I will fix it either way

fesch, 2015-11-24:
Just tried to overload the Frames.fillStatic and default to a Zoe, then added a car parameter to reloadTiming that call fillStatic with the same parameter. As the frame are anyway reloaded after the fields are being initialized, this does the trick.
But putting the CAR into the MainActivity would be the logical way to do, as all settings are there and this is the first class to be loaded anyway. So that would make sense ...

yoh-there, 2015-11-24:
I agree. Will do

yoh-there, 2015-11-24:
This should be better.

closed 2015-11-26 by yoh-there