fesch/CanZE #235: Tyre pressure status screen.

sordoneF, 2015-11-28:
I had a glitch where my Zoe popped a tyre pressure warning. But there is no indication where the problem is. I hooked up my i907 and in UCH channel 38 - 41 the tyre status are shown.
Front left tyre had "status5" where all other 3 had "status1". After I started my ZOE the warning was gone (??) but no tyre pressure adjustment was done.
It would be nice for non i907 owners to see the tyre status with a car and tyre problem (status <> 1) location icon in CANZE. I can perform testing if needed since my ZOE has TPMS. Can you get other statuses information ?
See http://heppsauto.com/wp-content/uploads/2014/10/tpms-tire-pressure-sensor.jpg for an example of an error screen.

yoh-there, 2015-11-28:
Thank you for the hint. I have a i907 on loan at the moment and can confirm the valve parameters in those positions. Unfortunately, that tells us nothing about the messages used, but tomorrow I will check "sniffing" the traffic on the CANbus when I pull those 4 fields. With a bit of luck there is something to learn. I do not have TPMS myself as my car is a 2013 build and I don't think it has the needed hardware. However, I do see the Status1 messages on the i907 so there is hope.

If I can dig some data from the bus I will make an experimental screen that you could test.

yoh-there, 2015-11-28:
Double (somewhat) with (closed) issue #175 . If you have any info on how the valves are activated (i.e. after changing wheels/tires) that would be helpful, especially if it is done using the standard CLIP tool or a separate tool.

yoh-there, 2015-11-28:
I found some info in the free flowing frames. Status info seems to correspond as follows:
0: pressure OK
1: no pressure info
5: Flat tyre (under 2nd threshold)
6: Under inflated (under 1st threshold)
So, I am a but confused about your ( @sordoneF ) status 1 messages. They correspond with mine and seems to indicate: no sensor, which I believe is correct for my car, but I would have expected 0 for yours. Unless of course your dealer screwed up........

fesch, 2015-11-28:
What frame are you talking about? A friend of mine called me today regarding the tire pressure ... so I think he will be interested to see this in CanZE too ;-)

yoh-there, 2015-11-28:
673 Free frame. I am adding it to the Gsheet right now

sordoneF, 2015-11-28:
I checked again and all four tyres are now at status1 which is odd. Status > 1 detection would be nice then. Next week tuesday I would get a set of wintertyres with Renault TPMS sensors in it so I can double check with the tyre fitter who has the TPMS tools. In two weeks my ZOE is having it's first annual service so I add this to the check list.
I definitely had no flat tyre so it is an odd hickup. Temperature was low at 0 degree C when it started and stopped 30 km later.

yoh-there, 2015-11-28:
I will add to experimental before I head to bed, so it can be included in tomorrows release

yoh-there, 2015-11-28:
Committed the tyres screen. Crappy layout and it doesn't work at all on my car (all data shows 0), so please report back once this is released.

sordoneF, 2015-11-29:
[image]
[image]
Two screenshots, second one was with car in sleep mode. When I started the car, the pressure values went to 18.6. In i907 the status is at "status1" for all 4.
Is the value in kPA ? I topped up my tyres with NO2 at 2.7 bar and NO2 tends to keep its pressure at all temps.
1 PSI = 6,894.75729 pascals.
If I divide 18,6/6,894 = 2.6979 which could be right in Bar but is PSI ?

yoh-there, 2015-11-29:
Thank you! The values SHOULD be in mbar, according to the documentation, so the values are weird. I will check again.

yoh-there, 2015-11-29:
Yeah, stupid factoring mistake on my end.

However, the 18.6 corresponds to "unavailable", not a real pressure value. I will ensure an unavailable is shown instead of a silly high value, but the pressure is obviously not there. Now I know these TPMS systems do not transmit very often so it could be that values are available after driving a couple of minutes. I am still trying to figure that out. Status 0 means: OK, so I wouldn't be too worried about that right now.

sordoneF, 2015-11-29:
[image]
[image]
I took my Zoe for a spin and after a few seconds, the pressure values changed from 18,6 to ca. 12,5. See pictures. It seems to be working but not in mBar. Odd. My tyres are not running at 12 mbar. :-)

yoh-there, 2015-11-29:
That is great actually, thank you, as it indicates the system is working. The wrong factor does that, you'd have to multiply it with 188.375 (don't ask) to get the mbars. I already corrected that in the source. 2411 mbar DOES make sense!

sordoneF, 2015-11-29:
Can you create a new build so I can check my new wintertyres coming next Tuesday with the mbar values ?

fesch, 2015-11-29:
try this one ...

fesch, 2015-11-29:
http://www.fisch.lu/junk/app-release.apk

yoh-there, 2015-11-29:
@sordoneF  As a return favour ;-) Could you interrogate the mechanic how they register the new tires? If it is solely done through the diagnostic connector, we might build that in CanZE, though what I mostly read an "activator device" (to put the sensors in "test" mode) is required. But maybe that is ALL that is required, I just don't know.

closed 2015-11-29 by yoh-there

sordoneF, 2015-11-30:
[image]
The tyre pressure screen works like a charm in both landscape and portrait mode. See above. It is a really nice addition for range optimization since the Zoe has no way to show this valuable data. Maybe add a "mbar" label field in the corners. On start, it shows 3499 as values and goes down once it gets it's real pressures. One issue: you can not exit the screen once you start driving. You are not allow to go back to the main screen while driving. And you can see the values only when you are driving. For example I adjusted my tyre pressure at a gas station and went on the highway to see the pressure values. Then I have to stop on the highway to exit to my driving screen in the app ? That does not make sense. And I have a co-pilot who can work the screen while I am driving. The app is too restrictive in this manner in my view. Tomorrow more info on the TPMS procedure.

yoh-there, 2015-11-30:
Thanks for testing. We'll promote it soon to the main or technical screen.

The 3499 I will investigate, it should also have a corresponding status like "not available" or something.

You can disable drive protection in the settings screen, it is called "Safe Driving Mode"