fesch/CanZE #249: Andoer ELM327 OBDII

ener73, 2015-12-07:
This OBD is not working with canZE. 
http://www.amazon.it/gp/product/B00F0H4QIK?psc=1&redirect=true&ref_=oh_aui_detailpage_o09_s00

The bluetooth terminal, sending "ATZ", gives this answer: "ELM327 v1.5?". So it seems the connection to the phone is given but no data is send. 
The message  "err ats0" appears in the CanZE app.

What is the problem with this OBD?
Thank you

yoh-there, 2015-12-07:
The rude answer is: it is not the recommended dongle. There is no way we can figure out what is wrong. There is a lot of information on the canze website about the mess these dongles are. All firmware is illegal and lying about there capabilities. Get a KONNWEI or MAXISCAN please. Links to ebay on the canze website.

closed 2015-12-07 by yoh-there

fesch, 2015-12-07:
If you want, you can send me your dongle and then I could "live" check it. There is no remote debugging method.

You are using a Zoé, are you?

ener73, 2015-12-11:
Yes I have the Zoe. I purchased now the Maxiscan that works fine. Thanks

yoh-there, 2015-12-11:
Thanks for reporting that back, it is appreciated. Have fun with CanZE.