fesch/CanZE #250: Is there any work for a programmer starter?

tsknightstorm, 2015-12-07:
Hi all
Sins 2 years i live with my lovely ZOE. But like all womans i never know whats going on in her. So I started to search a way to get more informations and i found your Project. If i am right there is a lot of work to do in this Project and the Programm for arduino isn't finish yet. My goal is to log different values with a arduiono until i drive and charge. Is there some Thing I can do as a realy starter in this Project? I would like to learn some Things about programming, can-bus and of corse about my is thinking ZOE.

This summer i started with a electrotechnic Ingenieur study. That means my focus is more on the electronic and on tests.

Thanks for your answers and sorry for my bad english.

fesch, 2015-12-07:
While waiting that I'l clean up my code and put it on Github, you may use this to analyse:

http://www.fisch.lu/junk/CAN-Reader-DUE-RR-cleaned.zip

fesch, 2015-12-07:
I'm marking this as "closed" as it is no really an issue. Feel free to continue writing messages here or mail me, which will probably be faster ;-)

closed 2015-12-07 by fesch