fesch/CanZE #251: immediate crash at tap on "Battery" - "Cell Voltages"

jragg, 2015-12-07:
immediate crash when tapping "Battery" - "Cell Voltages" 
Canze V1.07
Android 4.4.4
RedMi Note 4G

fesch, 2015-12-07:
Could you enable the "debug loggin" in the settings, restart CanZE, reproduce the crash and then send us the latest log file contained on the SD-card in the "canze" folder?

jragg, 2015-12-07:
Here are the last two files:
[2015-12-07-22-50-30.log.txt](https://github.com/fesch/CanZE/files/54611/2015-12-07-22-50-30.log.txt)
[2015-12-07-22-50-12.log.txt](https://github.com/fesch/CanZE/files/54612/2015-12-07-22-50-12.log.txt)

fesch, 2015-12-07:
Seams like there occurs a NullPointerException because an empty field is being called:

2015-12-07 22:50:25.408: !!! >> Field with following SID <> not found!

What car do you have selected in the settings?

jragg, 2015-12-07:
Car is ZOE in Settings and in real ;-)
Thanks!

yoh-there, 2015-12-07:
Same problem here. If I open "Battery", the widgets show, but as I tap on them, things crash. No definite pattern yet.

fesch, 2015-12-08:
Ahhh .. when double tapping on it. I now finally can confirm and reproduce...

Will take a look at it.

fesch, 2015-12-08:
@yoh-there: please check if the latest commit also works for you, so we could close this issue too ...

yoh-there, 2015-12-08:
No crashes :-)

closed 2015-12-08 by yoh-there