fesch/CanZE #258: Each Virtual field creates a new frame and ECU

yoh-there, 2015-12-12:
This is slightly illogical and uses a little bit of memory. Suggestion is to use the ResponseId (as used for ISO-TP) as the selector of the proper calculation and create only one virtual ECU and virtual Frame.

ecu = new Ecu("VirtualField computer", 9999, "", id, id, "VFC", "");

Frame creation needs no change.

VirtualField init parameter 1 definition should then be the reponsId (no ecu ID needed anymore).

Low prio.

yoh-there, 2015-12-20:
Changed this behaviour. Please note that now there is only ONE virtual Ecu (0x800) and ONE virtual Frame (0x800). Virtual fields are created indeed as fields in this Ecu/Frame with identifying responseId's, and thus mimicking ISO-TP frames.

At this moment, there are 3 virtual fields defined, and only one is actively used (consumption meter in ConsumptionActivity). I changed line 34 of activity_consumtpion.xml naturally.

closed 2015-12-20 by yoh-there

fesch, 2015-12-20:
saw that. Any reason why you put the ".24" behind?

yoh-there, 2015-12-20:
Yes. To mimick UDS frames/fields. Any number is arbitrary, so this seemed "closest to reality".