fesch/CanZE #260: Charging screen

solmoller, 2015-12-13:
Available Charging power is listed as #ACP 

Mean battery temperature is extracted as -40

jragg, 2015-12-14:
For me: It works ! right Power and temp data is shown.

Version CanZE: 1.08 
Car: Zoe
Dongle: MaxiScan 902
Smartphone: XIAOMI RedMi Note 4G
Android version: 4.4.4

yoh-there, 2015-12-14:
jragg, I think your car is a Zoe right? solmoller is a FLuence ZE

yoh-there, 2015-12-14:
@avataranedotas What frames do you use for AvailbleChargingPower (if you use that), meaning, what the external charger is willing to give (pilot signal on CP). I use 427.40 and that seems not to work for "FluKan". Same for men battery temp, I use 42e.44

If course I can make this Zoe/FluKan dependant, but I would need the proper frames.

jragg, 2015-12-14:
yes right, a ZOE from 2013

avataranedotas, 2015-12-14:
For the pilot signal charging power I get the Amperes from free frame 42E, bits 38 to 43.
427.40 is not available on my Fluence ZE.

Battery temperature is not available on free frames on my Fluence ZE.
I get it from the LBC requesting 0x022103, Byte 4 and Byte 5 (starting at Byte 0) of payload data with 0.1 C precision.

yoh-there, 2015-12-14:
Fixed #269

closed 2015-12-14 by yoh-there