fesch/CanZE #262: Formal Disclaimer answers not fully visable on phone for o.a. Safe Driving Mode setting.

sordoneF, 2015-12-13:
Expected result: "Yes" and "No" buttons.
Device is Motorola Moto-G, Android 5.1, Resolution = 720 x 1280 pixels.

[image]
[image]
[image]

[image]
[image]

fesch, 2015-12-13:
Hui, if this is 720 x 1280 resolution, then my actual strategy is bad. Need to check to get the dpi settings. Actually my 7'' pad has the same resolution, me the entire display is much smaller ...

sordoneF, 2015-12-18:
https://play.google.com/store/apps/details?id=com.MTMDevelopers.ScreenSpecs, this app will say: 720 x 1184, DPI = 320 which is more or less correct.
http://www.whatismyscreenresolution.com/ will say 320 x 640 which is not correct. Strange so be carefull and select the right test.

fesch, 2015-12-19:
I've tried to fix this by including the scaled density into my calculations. Hope it will work ...

fesch, 2015-12-22:
If anyone would like to test, the intermediate version is here:

http://www.fisch.lu/junk/app-release.apk

sordoneF, 2015-12-22:
Tested the 2015.12.22 version and the problem is not solved. Maybe a bit better but the text is not "full screen". Maybe change it to a "No" and "Yes" button ?

[image]
[image]

yoh-there, 2015-12-22:
Sorry, but noticing a few spelling errors too.

Turn off your (of), but for clarity I would suggest "even if you power cycle your"
interfere (interfer)
applications (application)
also may (alsomay)
drain your battery (suck of your battery)

fesch, 2015-12-22:
Actually I have a condition that put a simple "no" and "yes" on the buttons if the screen resolution is too not high enough to fit, but that does not seam to work.

I will implement some test-screen I will ask you to send me a screen capture so maybe that will help me get things ordered.

yoh-there, 2015-12-22:
So, the emulation in Studio is crappy??? That is pretty bad!

fesch, 2015-12-22:
I've uploaded a new version with a button "SCREEN TESTS" in the experimental section. Please post a screenshot of that one ... and hopefully I will get some good information to base my test on.

http://www.fisch.lu/junk/app-release.apk

@yoh-there : in the emulator everything works fine! @sordoneF's device has a resolution of 720 x 1280 (around the same as my Nexus 7) but everything seams to be scaled up :-/

sordoneF, 2015-12-22:
Screentest done with as result:
[image]

yoh-there, 2015-12-22:
And for my S3

Opening screen
[image]

test screen
[image]

fesch, 2015-12-23:
I notice you both have a density of 2. Mine has a density of 1, respectively of 1.33 on my 21' tablet.

I've modified my test:
http://www.fisch.lu/junk/app-release.apk

sordoneF, 2015-12-23:
Display is now  correct. Text "Do you agree" should be added as last line, I think. Safe driving is stille off.

Greetz frank/sordonef.

yoh-there, 2015-12-23:
[image]
[image]

fesch, 2015-12-23:
OK, we need to shorten the text  of make it smaller ;-)

yoh-there, 2015-12-23:
Why? I didn't try but it is scrollable isn't it? Nobody reads it anyway.

fesch, 2015-12-23:
OK, so then this is fixed :-)

closed 2015-12-23 by fesch