fesch/CanZE #265: Bluetooth/Connection problem

Zoelibat, 2015-12-13:
Version CanZE: 1.08
Car: Zoe
Dongle: MaxiScan 902
Tablet: Gigaset QV830
Android version: 4.2.2

I can get a bluetooth connection to the dongle (dongle is paired and blue light is on) and CanZE shows "connected to maxiscan". But i can't get any data from the car. When i choose the battery page it shows NaN and no real data.

[2015-12-13-21-19-14.log.zip](https://github.com/fesch/CanZE/files/60604/2015-12-13-21-19-14.log.zip)

fesch, 2015-12-13:
As far as I can read from the logs, CanZE tries to open up a second BT connection which is then interfering with the one that already has been set up.

I hoped to have eraased this problem (it did not again happen to me), but unfortunately I didn't :-(

You could try to power off the dongle, close CanZE, power on the dongle and start CanZE again. Maybe cycle through this 3-4 times. 

When starting CanZE, give it some time to settle up the connection ...

Zoelibat, 2015-12-13:
Nothing changed.

See attached files.
[CanZE.zip](https://github.com/fesch/CanZE/files/60612/CanZE.zip)

[image]
[image]
[image]

fesch, 2015-12-13:
Does the toast "ELM is ready" appear on the screen?

Your logs contain data for maximum 3 (!) seconds...

Will take a deeper look at it tomorrow. For now I still think you are acting too fast.

Zoelibat, 2015-12-13:
"ELM is ready" does not appear on the screen. What means "toast"?

Acting too fast? How long do i have to wait?

fesch, 2015-12-14:
A "toast" is a small message that appears at the bottom:
http://developer.android.com/images/toast.png

It needs up to 5 seconds if everything goes well ...

Zoelibat, 2015-12-14:
I was waiting at least half a minute after starting CanZE.

In my opinion it is a bluetooth bug in Android 4.2.2.

fesch, 2015-12-14:
Possible. I think I can get a 4.2.2. device to test ...

jragg, 2015-12-14:
A similar problem here:
Version CanZE: 1.08   (same in v 1.07)
Car: Zoe
Dongle: MaxiScan 902
Smartphone: XIAOMI RedMi Note 4G
Android version: 4.4.4

When starting app and connect the dongle everthing works for abt 3-4 minutes. Then bluetooth freezes and nothing happen in the app screen...
After going back to Android Main Screen and then shutdown CanZe with the taskmanager and then restart then Canze is working great for at least on hour ! (not longer tested) 
I must really shutdown canze with taskmanager, only going back to Android Main Screen and restart CanZe  again does not help.

fesch, 2015-12-14:
OK, then I think it's not a BT problem but rather something that has to do with the dongle.

@yoh-there : I think we should try lower the ELM327 settings again ... yet I don't not know how.

yoh-there, 2015-12-14:
@fesch as in? More lenient timeout for free frames?

fesch, 2015-12-14:
That or maybe out adaptive algorithm, which seams to work fine, is not "stable". As far as I remember, the timeout is being increase dynamically and also decreased dynamically.
So if we hit the "limit", the algorithm enter some state where it increases, then decreases, then increases, then decreases .... the timeout.  Each decrease will be followed again by a timeout (longer wait, source of unstability, ...).

Do you see what I mean?

Zoelibat, 2015-12-14:
Today i used a galaxy S5 (or S6?) and the toast "ELM ready" was shown. But after 10 seconds the toast "ELM not responding" is shown.
Now i am completely confused. I thought on a newer Android i will get it to work. Unfortunately i have no logs.

Bluetooth issue on Android 4.2.2.: https://code.google.com/p/android/issues/detail?id=24522

Maybe the dongle is the problem? I don't know.

yoh-there, 2015-12-14:
OK yes, however there is I would think enough reason to stay dynamic.

Increase is fast, 0.1 each error, decrease is slow, 0.01 every success and both ends have "end stops": minimum factor is 1, maximum 2. This is all happening around line 700 in Class Device.

As a temporary measure I would suggest to up the minimum to 1.3?

yoh-there, 2015-12-14:
OK, set dynamic timeout increase to 1.3 minimum value. #268  This might help, though I have not seen this being an issue on Zoe. Never say never.

fesch, 2015-12-20:
@Zoelibat : did the changes bring something positive?

Zoelibat, 2015-12-20:
Is there a new version out? Did i miss something?

yoh-there, 2015-12-20:
I guess you DID miss something ;-) http://canze.fisch.lu/release-schedule-change/

And I think those changes were not release on google play yet, though I cannot vow for that, and the comment more or less implied they did.

ps: I run Android 4.4.2 on my S3. If there is anything specific I can do?

Zoelibat, 2015-12-21:
Maybe i'm blind but i can't see a new version. I will compile it by myself.

Any good tutorials? Anything to know about CanZE and android programming?

yoh-there, 2015-12-21:
I just meant there is no fixed every Sunday release anymore. 

On github there is also a wiki. Look at the right hand side. It contains details on how to install, fork and compile.

Zoelibat, 2015-12-22:
What does "LINSCH" mean? The log file shows me "Ecu does not exist: LINSCH".

yoh-there, 2015-12-22:
LINSCH is the name of an ECU (computer in the car) that does not exist. You can happily ignore that error.

fesch, 2015-12-23:
Yesterday, during a longer drive and even later letting the tablet connected an on during the charging session, the activity closed unexpectedly so that the app came back to the main activity with the buttons.

Has anyone already had the same symptoms?

Zoelibat, 2015-12-23:
Sometimes i get the disclaimer twice. 

I have tried it with the new (unofficial) version but with the same result. I can't get CanZE to talk with me. :-(

I've bought another tablet. I want this thing running. Dear Santa Claus, i wish my dream come true. ;-)

Happy Christmas everyone, great work! Thanks!

yoh-there, 2015-12-23:
@fesch Yes, on some "crashes", it crashes to the main screen. How don't ask me if the queue is still running. From memory yes, mostly things are pretty haywire when this happens.

@Zoelibat We'll get this running on your car. Sure CanZE has it quirks, but we're getting better at it. Thanks for your reporting.

yoh-there, 2016-01-06:
I am adding @zmeglic and @tsknightstorm in this post so they will receive updates on this thread when they are made.

GlowPod, 2016-02-10:
Version CanZE: 1.09
Car: Kangoo
Dongle: OBDLink MX
Smartphone: Samsung Galaxy S5
Android version: 5.0

Another dongle that read any data as others describe, graphs show NaN. The dongle has an ELM327 v1.3a so it should work according to notes on the CAN ZE website. I have verified that the phone and dongle work perfectly on a Mini so it's not a hardware/phone issue.

A message briefly pops up saying unrecognised command or response. I didn't manage to catch the details but I can try again with the log later today.

Any suggestions?

solmoller, 2016-02-12:
Please clarify where you see that dongle as supported

Br

Henrik

yoh-there, 2016-03-27:
I noticed when things get stuck, also my phone interface on R-Link was dead, so that hints in my view to a problem in the phone, not in the dongle. Of course unless the dongle would be so confused it was saturating the channel.

So, after that I have taken a different approach and instead of starting/stopping CanZe and/or the dongle, I now stop and start the Bluetooth interface on the phone, without restarting the dongle or CanZE. Until now, this has ALWAYS and fairly quickly restored operation. Maybe it is a hint of what's going on.

fesch, 2016-03-27:
I have an iPhone for the BT connection to the car. I will switch of that next time an give it a try ...

yoh-there, 2016-07-11:
Got a new phone Friday, a Cubot X15 with Android 5.1. It's really too early to tell, but today I drove 90 km's without a hitch. Would the problem be the phone / OS after all? Hmmmm. Wil update this comment if lockups still occur.

fesch, 2017-01-07:
Got no updates here --> closing

closed 2017-01-07 by fesch