fesch/CanZE #271: In battery screen no cell voltages are displayed

macfrank, 2015-12-17:
In battery screen no cell voltages are displayed. The chart in the left lower corner is empty. 
[image]

fesch, 2015-12-17:
Hi,

what dongle do you use? What car do you drive?

macfrank, 2015-12-17:
My dongle is a MaxiScan902 and the car is a Renault Zoe R210

fesch, 2015-12-17:
Same as mine thus. Strange ...

@yoh-there : What do you think of this?

yoh-there, 2015-12-17:
@fesch Actually, on mine too........ But the corresponding voltage heatmap works. So my guess, without looking at the code is: config error or some "overloading". Assuming it works on your tablet, could it have to do with the widget being "too small"?

yoh-there, 2015-12-17:
Oh, I realize macfrank has a tablet. So it is NOT the size. Well, as I have the problem too and you @fesch do not (right??) I think I should investigate.

fesch, 2015-12-18:
I will check this morning that screen. Haven't been in there for a while ...

fesch, 2015-12-18:
I can confirm this: no data displayed! I will do a deeper analysis, as I did not touch that screen for some time.

fesch, 2015-12-19:
Fixed

closed 2015-12-19 by fesch