fesch/CanZE #273: Data streaming stops sometimes because bt connection is lost and reestablished after some time

macfrank, 2015-12-17:
streaming stops sometimes because bt connection is lost and reestablished after some time.
It also seems that some data is not refreshed periodically. 

[image]

fesch, 2015-12-17:
Reading your graphs I conclude that you have serious communication problems between your Android device and your dongle. Please give more precise information about both devices ...

macfrank, 2015-12-17:
My dongle is a MaxiScan902 bt. The screenshot is from a cheap 7" tablet (Odys Connect PRO 7) which runs with Android 4.4 (KitKat). But I encountered the same behavior with a Samsung Galaxy S2 running Android 4.1.2.

macfrank, 2015-12-17:
May be the bt connection is weak because I've always connected my iPhone with bt to R-Link and this interferes with the bt connection between the MaxiScan dongle and the Android device?

fesch, 2015-12-17:
Maybe, but I have my BT connected phone (with R-Lnk) laying right near the dongle too ... :-?

macfrank, 2015-12-17:
Ok, this is what I would expect too. Shouldn't be a problem with two bt devices in the car. I did some further investigation in the car. Mostly the bt connection is not dropped completely because the blue led from the dongle is still on. But the green leds stop flickering and no data are captured anymore in CanZE. The situation that bt connection is dropped completely (blue led turns off) are not often.

yoh-there, 2015-12-17:
Well, I would at least try to do a drive without the phone's BT on. My setup is: phone does BT to R-Link AND to the dongle at the same time, but then the phone (an S3) is a Bluetooth master device.

I know two completely separate channels should work, but "should" and "I assume" are usually the mothers of all F****-ups. At least give it a try.

Having said that, it is my experience that sometimes the link breaks and often CanZE can fix it, and sometimes........ not. We (well, fesch) has spend many, many days on that and I have the impression we can't improve much further. Sometimes it seems it is the dongle that ends up in la-la-land.

macfrank, 2015-12-17:
Ok, tomorrow I'll give it a try without the iPhone bt enabled. I'll report if the behavior changes.

macfrank, 2015-12-17:
Some more charts. This one with heavy bt issue. 
[image]

And these after restoring the bt connection but one with error message. But bt seems to work and data is streamed all the time.

[image]

[image]

[image]

yoh-there, 2015-12-17:
Nice graphs :-)

Did you notice this happening, and if yes, did you have the impression there was traffic going on with the dongle (un-regular flashing LED's)?

And were there any of those toast messages (those black small popups) while the thing had lost it's mind?

macfrank, 2015-12-18:
Honesty I don't understand your question.
When data streaming stops the blue dongle led keeps on but no traffic assigned with the green dongle leds. In the first screenshot of my last post, the blue led turned off too. The bt connection was completely lost.

yoh-there, 2015-12-18:
OK, that is what I meant, no "traffic" (green flashing LED's on the dongle).

Maybe you didn't understand what I meant with a "toast". It is what shows in the second image in your 6th message here in this issue, the black box with "Err atfsm.....". That black pop up is called a "toast" in Android :-) Don't ask me why!!!

[image]

I had a hangup in my 40 km drive this morning like this (note to self: I noticed it at 08:18). I had to restart canze AND the dongle. First tried to power cycle the dongle only, but CanZE did not recover. After that I stopped (not killed, just stopped, but I do NOT have the run in background on) canze and started it again, and it ran again. A bit of a pain in the behind and I wish we could find a way to around it. @fesch I am now at a client, but when I am back home I will dig up the logs and send them to you.

macfrank, 2015-12-18:
:-) Ok, many thanks. Now I know what's a toast....
I did two tests today. First test was with iPhone bt on and connected to R-Link. Second test was with iPhone bt always off and not connected to R-Link. From my point of view this did not have any influence for the bt behavior of CanZE with the dongle. Even with iPhone bt off I encountered the same issues than before. 
A really strange thing is, that the toasts were shown even if bt connection stays stable and green leds keep flickering. But no data has been streamed or shown in the graph/charts. My impression is that the bt problem is caused by CanZE because most of the time when bt connection is broken, the problem is solved by killing CanZE and restarting it. Most of the time there is no need to turn dongle off/on.

macfrank, 2015-12-18:
Screenshots from testing results with iPhone bt on:

Toast but blue bt led on and green leds flickering.

[image]

Seems that data streaming is not affected at all.

[image]

fesch, 2015-12-18:
The toasts are "delayed", so whatever has happened is already over when the toast appears ...

yoh-there, 2015-12-18:
Thanks for that feedback! Good to know it is definitely not the other BT connection that is screwing things up, so we'll erase that from our list. Actually, I do have another new idea what could cause this and we will look into this idea soon. It is NOT something unique to your setup, so don't despair and you don't need new hardware by the looks of it.

The now-and-then toasts you can safely ignore as long as things keep moving along. It is "normal" that now and then a data-point is missed. You can get rid of the annoying toasts by setting the log level to "None" in the settings screen.

macfrank, 2015-12-18:
Now testing with iPhone bt off. But serious problems in CanZE.

[image]

[image]

[image]

[image]

[image]

[image]

Now charging the Zoe and still toasts. 

[image]

fesch, 2015-12-22:
Duplicated: Bluetooth/Connection problem #265 

--> Closing this one a I'm on that anyway ;-)

closed 2015-12-22 by fesch