fesch/CanZE #284: bluetooth connection error

schouf26, 2015-12-24:
hi,
i'll be try to explain my problem with my poor english.
i used canze since canze 0.9 with my LG G4 (under android 5.1), without problem, in october i had lot of problem (near canze 1.01 or 1.02), messages on my screen ( err ca[], disconnect, lot of number, try hard reset, many more), nothing has worked, i try to desinstall and reinstall canze (old version and new version), nothing to do, can ze won't work.
i thought it was my dongle KO, i bought a new one, and same problem. i change my phone with new one too, same problem, i try to upgrade my phone to android 6.0 ....same problem.
i try my son's phone (moto E android 5.1) and everything is OK but don't work on my asus tf701 (android 4.4).
I don't try the old dongle with my son's phone, i'll do it this afternoon.
i put screenshot and log later (because my wife has stolen my car this morning). 

thanks a lot for your work.

yoh-there, 2015-12-24:
Dear schouf26.

1) Have you installed the most recent version too (1.07 right now?) We have had problems indeed, but we feel most of them are solved ad that is left is being addressed in #265 . If you have not, please do so first (if you install from Google Play, it will automatically update anyway).

2) I assume you are driving a Zoe. Is that correct?

If you have the most recent version and driving a Zoe, please set log level to "None" in the setting screen and see if CanZE performs for you.

Thanks

schouf26, 2015-12-29:
dear Yo-there,
i'm so confuse, i try witout log level, and everything is ok, but acquisition was slowly than before (for me), i don't try canze 1.09 (not yet).
yes,  i am zoe driver.
thank you.
can you tell me  why log level was blocking my canze, please ?

yoh-there, 2015-12-29:
What could have happened that things got messed up and a lot of error messages (those black popups are called "toasts") were generated. Android queues those up and displays each for two seconds, so you can imagine things become totally uncontrollable if 50 messages are generated.

Normally, things go wrong now and then, but CanZE recovers from it. If you set the log level to none, these "toasts" are simply not displayed, and if things work fine, I would recommend to leave it there. (I do!)

I assume you have a Konnwei or Maxiscan, so things should basically just work.

I am closing this issue, but if you feel there is a serious problem left, just reopen it.

closed 2015-12-29 by yoh-there

sven1555, 2017-04-04:
I'm also expieriencing problems with my LG G4 (Android 6.0). The connection to my ZOE Q210 via CanZE through the ELM327 adaptor only works for a few seconds. For example I'm able to view the voltage heatmap once but after that the communication got stuck. There is no refresh and also after changing the view there is no data read and displayed. Tried to set the log level to "none" but that doesn't change the behaviour except of not showing the toasts. 

My OBDII-adapter is alright, since two other android devices just work fine with the adapter and CanZE.
- Samsung Galaxy S2 (Android 4.1.2)
- Acer Iconia Tab A700 (Android 4.1.1)

But as you can mention, I don't like to take any of my vintage devices with me just to get CanZE running. 

As a hint: The debug infos at the bottom row in CanZE are changing quite faster with the LG G4 comparing to the other devices. May the issue be caused by polling the data too fast / don't wait long enough for the answer on that faster phone?