fesch/CanZE #291: Err atfcsh77b with ELM322 1

Nono1200, 2015-12-28:


Nono1200, 2015-12-28:
Hello,
I own a recently ELM327 1.5. It connects correctly Bluetooth, but it connects it displays this error err atfcsh77b.
Do you have information for me?
Thank you

Arnaud

yoh-there, 2015-12-28:
If that is incidental, please ignore (you can switch those messages off setting log level to none in the setting screen).

If it is NOT incidental but always, please state your device or include a picture. Thank you

Nono1200, 2015-12-29:
It's always and this prevents the proper functioning of the case.
Here are photos:
[image]
[image]
Thank you

yoh-there, 2015-12-29:
OK, well here is the problem. The fcsh command is absolutely needed for the operation in this EV, and obviously, the dongle does not support it. Other than the Konnwei KW902 and the Maxiscan we cannot help with other dongles. The one you have is totally no-brand in a generic case. In fact, I own EXACTLY one looking like this and it works fine. You just can't tell what is inside.

We have tried to explain this as clear as we could here http://canze.fisch.lu/elm327/ 

>The cheapest knock-offs do not work. Avoid the tiny ones in a transparent blue shell, as
> most contain stripped logic cramped in the spare room of their Bluetooth controller and
> these are too limited to work for CanZE.
> It is somewhat slow, though we’ve been able to get more out of it than we ever expected.
> If you experience problems with dongles other than the KW902, we are unable to help.
> The KW902 works, it is relatively cheap, and it is simply not worth, if not impossible, to
> cross check 50 unbranded, unversioned knock off dongles against 75 Android devices.

While I would really like to say differently, there is absolutely nothing we can do for you, other than advise you to buy a Konnwei or Maxiscan. They are available on eBay for under 15 euro's from China (it will take 2-3 weeks to arrive) and under 25 euro's from Europe (usually Germany)

closed 2015-12-29 by yoh-there

Nono1200, 2015-12-29:
OK thank you.
Can you give me a link to ebay dongle Konnwei KW902 I am sure not to be wrong please?
Thank you very much.
Arnaud

yoh-there, 2015-12-29:
You're welcome. There are three example links on the Hardware page of the CanZE website.