fesch/CanZE #297: Request: colour change

fr0d0scious, 2016-01-01:
The white background is quite distracting unless you dim the display right down.. then the data becomes difficult to read.

Would it be technically possible to have a black background option? (to match the in car display colours)

Many thanks

fesch, 2016-01-01:
technically, yes ...

yoh-there, 2016-01-01:
It is an awful lot of work to do that, not just dropping in a new theme. See issue #218 . I assume you're talking about night driving primarily and would suggest to read that issue thread first. I am closing this issue under that assumption. It can always be reopened.

closed 2016-01-01 by yoh-there

fr0d0scious, 2016-01-02:
It's all the time I was asking about. I wouldn't want a sensor mode change.

Reading that linked thread it occurred to me that I'd just removed the invert colours option from the quick access tiles on my one plus one (it running Cyanogen 12). I tried that and CanZE looks great! So my problem is solved.

If you ever need help with colouring odd be glad to help.

Thanks for your time.

yoh-there, 2016-01-02:
Glad that is "solved" ;-) I could not find anything that could do recoloring for an unrooted phone, other than "dimming" skins and adding a bit of color. My S3 is, realistically, too small for the graphs to be readable anyway while driving and that is not a bad thing per se.