fesch/CanZE #298: Communication stops

zmeglic, 2016-01-04:
Version 1.09 and previous
Zoe
KONWNWEI OBDII http://www.ebay.co.uk/itm/151753972025
Nexus 5, up to date.

After a random time (a few seconds up to a few minutes) any screen simply stops updating.
The green communication lights on OBD stops blinking.
After returning to main screen, and selecting another one, the communications start again.
There is no error or toast message displayed.

tsknightstorm, 2016-01-04:
Same Problem here. ZOE and HTC One X+ with the Konwnei Dongle

fesch, 2016-01-04:
Know problem of the dongle. Working to improve that since September 2015. 

Actually you have to live with that until a solution is found.  - Sorry -

closed 2016-01-04 by fesch

zmeglic, 2016-01-04:
Good to hear there is still hope :)