fesch/CanZE #301: Tapping widgets

yoh-there, 2016-01-07:
A German user reported that tapping widgets leads to unpredictable results. Sometimes the single, big widget appears, sometimes nothing happens, sometimes the app crashes.

I can confirm the "nothing happens" (S3). I never used this feature myself so that's why this might pop up so late.

fesch, 2016-01-07:
When you touch a widget, it open at full-screen.

When you tap it but drag the finger into some direction (slide), nothing happens, because I reserved this sliding movement for some later feature ... not yet implemented.

If the app crashed, there mus be some other problem ...

yoh-there, 2016-01-07:
I can assure you either this is not (always) true about tapping, or my phones tap-versus slide behaviour is way way way off (it isn't ;-) )

yoh-there, 2016-01-07:
Maybe this helps. When I tap a widget in "Consumption":
01-07 23:43:10.511 5080-5080/lu.fisch.canze D/  CanZE: WidgetView: maskedAction = 0
01-07 23:43:10.571 5080-5080/lu.fisch.canze D/  CanZE: WidgetView: maskedAction = 2
01-07 23:43:10.571 5080-5080/lu.fisch.canze D/  CanZE: WidgetView: maskedAction = 1

yoh-there, 2016-01-07:
Yes. I looked at your code. There are many many times "2" (move) events even when I tap very, very, very light

yoh-there, 2016-01-07:
In WidgetView.java I commented out line 263, problem solved. Well, for now. It seems the slide trigger is way too sensitive. I think the move should only be triggered if it's position has at least once been past a trigger distance from the down event.

yoh-there, 2016-01-07:
Fixed in pull #302

closed 2016-01-07 by yoh-there