fesch/CanZE #310: Widgets sometimes display a huge triangle

yoh-there, 2016-02-05:
No big issue. Graphs now and then display a filled area under a line going from left top to right bottom. It disappear after the next update, so it's usually a "flash". Hardware S3.

fesch, 2016-02-06:
I can confirm this, but did not yet find the reason ...

yoh-there, 2016-03-27:
Seems to happen when there is a "gap" in the recording. Not 100% sure

fesch, 2016-04-18:
I did not get this since a long time anymore ...

yoh-there, 2016-04-18:
Sorry my friend, I do.

fesch, 2016-05-11:
Any improvements since my last commit?

yoh-there, 2016-05-11:
Haven driven longer stretches with a running graph in a while. I will
update today and drive tomorrow.

yoh-there, 2016-05-14:
I have seen one in my Thursday 40 km drive, but not down. Of course the flashes are very short and I didn't have my eyes on the phone all the time. Sorry. It's not a big deal (to me), but I think it's still there.

yoh-there, 2016-06-03:
Today, for the first time, I saw the "triangle problem" in the lowest graph of the consumption screen, which has a slow update rate AND draws dots. I was too late to make a screen shot, but it was very obvious the erratic line was drawn from the one-but-last time stamp to the left-top point. I hope this helps a bit.

Seems like there is "simply" a (0,0) value inserted. If anything else fails, maybe such values in the array can simply be dismissed?

fesch, 2017-01-07:
I did not get this since a long time. What about the other users?

yoh-there, 2017-01-07:
I propose to close pre-emptively

yoh-there, 2017-02-02:
It's still there. Not often and not a major issues. Just weird.

yoh-there, 2017-03-02:
Update: the wrong line is one of (too fast to see) going from top-left to the most current value, or the one-but-most-current value is interpreted as top-left. Is a (crude) option to just ignore data points that end up left-top (I assume that is 0,0 in the canvas)?

yoh-there, 2017-03-05:
Sorry, didn't fix it. I might have read your update wrong. Seems like your only checking "last" and not the m values? Shouldn't the strategy be: "if the m values are wrong, don't do anything, not even updating the last values?"

fesch, 2017-12-17:
Version 1.26 should now finally (and hopefully) fix this.

closed 2017-12-17 by fesch