fesch/CanZE #316: ENH: Temperature logging screen.

sordoneF, 2016-02-11:
Some Zoe drivers experience a switching of the internal temperature without changing the temperature settings. Can we have as an enhancement (please change this issue) a screen with a logging graph of the internal temperature, the temp. sensors and the temperature settings in a new "Thermo" screen.
X-axis: 1 or max 2 hours, ticks per 10 minutes ?
Y-axis: 0 to 30 degree C or selectable ?

yoh-there, 2016-02-14:
I have found the temp control setting and the external temperature, but not the measured cabin temperature. There is something that could be it, but I have to dig a bit deeper.

yoh-there, 2016-03-06:
Haven't had any luck with the i907 to find the internal temperatures. The issue is not forgotten, but slightly on the back-burner due to lack of that one resource that is both unlimited as well as always in short supply: time.

yoh-there, 2016-12-14:
Seems like the cabin temperature is CLIMA computer diagnostic command 2121, startbit 26, length 10 bits (so 0 - 1023 range), resolution 0.1C (so 0 - 102.3C range), offset 40 (so -40 - 62.3C range). Need to test this.

yoh-there, 2016-12-19:
I am closing this issue as it has been programmed in the experimental CLIMA screen. It has not been released (or even committed) yet, but it will make it's way. This is NOT the upcoming diagnosics value dump, but a neat screen with a 3 line graph: setpoint, inside, outside in a -10 - +40 range (yes @sordoneF , I expanded that a bit for the more extreme climates).

Time (and bug reports) will tell if this will move up to the main menu, but I think it will.

closed 2016-12-19 by yoh-there