fesch/CanZE #318: Crash at application start CanZE version 1.10

sapcmc, 2016-02-15:
Crash at application start CanZE version 1.10

Android version 5.1.1
mobile: One Plus One
bluetooth on

fesch, 2016-02-15:
The error code is here:

java.lang.NullPointerException: Attempt to invoke virtual method 'void
lu.fisch.canze.actors.Field.addListener(lu.fisch.canze.interfaces.FieldListener)'
on a null object reference
at lu.fisch.canze.actors.VirtualField.<init>(VirtualField.java:33)
at lu.fisch.canze.actors.Fields.addVirtualFieldUsage(Fields.java:114)
at lu.fisch.canze.actors.Fields.addVirtualFields(Fields.java:93)
at lu.fisch.canze.actors.Fields.load(Fields.java:1561)
at
lu.fisch.canze.activities.MainActivity.loadSettings(MainActivity.java:263)
at
lu.fisch.canze.activities.MainActivity.reloadBluetooth(MainActivity.java:614)
at
lu.fisch.canze.activities.MainActivity.reloadBluetooth(MainActivity.java:607)
at lu.fisch.canze.activities.MainActivity$8.run(MainActivity.java:526)
at java.lang.Thread.run(Thread.java:818)

=> What car did you select please?

sapcmc, 2016-02-15:
Hello,

Does not even allow me to select car. It crashes at initialization.

Best Regards,
Carlos Carvalho

fesch, 2016-02-15:
Did you have installed CanZE before?

sapcmc, 2016-02-15:
No, first time. Also I have cleared data and cache after crash and still
crashes. Be aware dongle is not turned on, just the mobile.

fesch, 2016-02-15:
@yoh-there : As far as I seen, the load method is being called and when no car is selected, the required fields will no be loaded into the memory, which is the cause of the crash I think.

sapcmc, 2016-02-15:
Hello,

So will wait for correction, thank you in advance.

fesch, 2016-02-15:
@sapcmc : if you can't wait, please try to download and install version 1.09 from the release section

sapcmc, 2016-02-15:
I can wait, it is fine, still don’t have the ZOE was testing application.

yoh-there, 2016-02-15:
Ah, that makes sense. I will have a peek. Worst case we quick-fix-default
to a Zoe right?

fesch, 2016-02-15:
Yes, that would make sense, as most use this setting I think ...

fesch, 2016-02-15:
fixed, new version uploaded to Google Play. Waiting for validation ...

closed 2016-02-15 by fesch

yoh-there, 2016-02-15:
I guess you agreed with my changed? ;-)

fesch, 2016-02-15:
Yes ... and first of all they worked. I hope to get back with more time soon ... still having some ideas to implement ;-)