fesch/CanZE #321: Settings not saving - version 1.11

sapcmc, 2016-02-16:
I have now the latest CanZE version.

If I go to settings and try to change them they do not save even if I press the green OK button.

I untick the safe mode, a pop up tells me I should not use it I say I want to then I click on OK button.

Back to main screen. Click on Battery, tells me  that I need to ajust settings first. I go to settings and the safe mode is still ticked.

Attached images.

[image]
[image]
[image]

fesch, 2016-02-16:
I don't get this bug reproduced. @yoh-there , what about you?

sapcmc, 2016-02-16:
I have a clean install if it helps

sapcmc, 2016-02-16:
To further add it also does not allow to change the defaulted car ZOÉ . If
changing to fluence as an example and pressing OK and return to main menu
if I go to settings again nothing is saved.

I believe issue to be in method that saves in database (that is not saving
but defaulting to ZOÉ in application without commiting values to SQLITE).

yoh-there, 2016-02-16:
I have removed the app (installed through studio), reinstalled through Google Play, went through the disclaimer hoops, set the car to Fluence and saved using the green OK mark. All settings were saved. So no, I cannot replicate this.

sapcmc, 2016-02-16:
Uninstalled app, installed again problem persists.

What data do you need from me? It does not save any settings on the
settings window.

This on a oneplus one with android 5.1.1

sapcmc, 2016-02-16:
maybe dumb question but after changing settings have you checked again in
settings if they actually changed? When I press OK and return from main
screen back
to settings it all defaults to the default settings without anything being
saved.

yoh-there, 2016-02-16:
I did......
1. start app
2. Yes on disclaimer
3. settings icon (spanner)
4. removed safe driving mode and confirmed
5. changed car to fluence
6. green save icon
7. settings icon
8. confirmed changes made
9. red cancel icon
10. close app
11. even close all cached apps (hold center button, clear all apps button)
12. start app
13. settings icon
14. confirmed changes made

yoh-there, 2016-02-16:
Had a friend check the exact above sequence with a oneplus 5.1.1. Same result as mine (aka successful save). I don't know what to say.....

sapcmc, 2016-02-16:
Video available at https://youtu.be/DbnIqpE0eKg so you can see it does not
work at least for me.

yoh-there, 2016-02-16:
Thanks. Back to @fesch ? I have no clue to be honest, sorry

sapcmc, 2016-02-16:
I believe issue is that it is not saving in database. It has the default
data when it launches but probably a more robust solution is:

   - when database is initial or purged calling settings at initialization
   of application so that settings are stored in database

This should fix issues.

sapcmc, 2016-02-28:
To further add  this is a bug for sure. I have tested now with a iOcean G7 with Android 4 and same thing not able to save settings.

You guys should look into this when possible.

yoh-there, 2016-03-01:
It's something we would really like to fix. We acknowledge your video really (thank you for the proof, it is appreciated). The other side of the coin is that we have been honestly unable to replicate the problem and have had no other reports about this issue. The code for save and retrieving is very straightforward. No activity on this issues doesn't mean we're not concerned. It means we're totally in the dark what 's going on on your phones.

Until this moment I can only guess that you have the habit to install "something else" on your phones, i.e. a virus scanner or whatever that interferes with the normal operation of the database calls, probably in combination with something minor in our code. Don't get me wrong, it's not an dismissal or accusation in any sort of way, it's simply that I can't think of an other direction at this moment.

sapcmc, 2016-03-01:
Yes, I am using a anti virus.

yoh-there, 2016-03-01:
Could you try, as an experiment, to switch if off? Again, that is not to shift the blame, but to make sure we know what's going on.

sapcmc, 2016-03-01:
Hello,

I am using CM Security. Did a force stop and same problem.

fesch, 2016-03-11:
Seams to be that this is CM related? But I did not get any freedback from other about this.

You may try to catch some logs and post them here ...

sapcmc, 2016-03-22:
How can I catch the logs? As mentioned the Iocean G7 has no cyanogen and same problem.

fesch, 2016-04-18:
Can I close this issue?

sapcmc, 2016-04-18:
Hi,

This still happens to me with android 4.4 on a iocean and lollipop cyanogen
on oneplusone.

sapcmc, 2016-05-14:
Issue can be closed. No longer replicable with latest cyanogenmod version.

closed 2016-05-14 by sapcmc