fesch/CanZE #327: Direct BMS connection

gcshave, 2016-02-29:
I am trying to read battery temperature and cell voltages for a Renault Fluence (Better Place) battery pack. I experimented with connecting the BMS CAN pins to the KW902 OBD2 bluetooth. However, no data transfers. CanZe software suggests that the car needs to be turned on.

Is it possible to use your software for this application (connecting directly to the BMS)? If so, how would I trigger the BMS to send data over the CAN bus?

Thanks.

yoh-there, 2016-02-29:
Yes and no. I will answer more detailed tomorrow. On the road now.

yoh-there, 2016-03-01:
OK, let's start first by saying my personal experience is the Zoe, not the Fluence. I am including here  reference to @avataranedotas as he is the Fluence specialist!

The bus you're talking about is called by Renault the Electro bus (the "normal" bus is called the Vehicle bus). I have a drawing how it is wired in the Zoe, but that is not very interesting. Just a bus with the LBC and the BCB on it, gatewayed through the EVC to the V bus.

We have next to zero information about the data content on the E-bus, which is very unfortunate. The only information we have are a few packets that are gatewayed to the V bus and that we know are coming from the devices on the E bus. What I am trying to say is: the little data that we have is 100% also available on the V bus.

Can the ELM device be used on the E bus? Sure it can, it's just a CAN bus. Just make sure you have the polarity of the two wires right and the grounding.

Can it be done with CanZE? No. We simply expect a certain bus speed (500 kps for the V bus) and some devices to be present.

Can CanZE be used as a basis to do that? Sure! All the (complicated) handling of the ELM, user interface, timer and queueing is already there and it would save a ton of work to start with the source code of CanZE. But you would have to know the bus speed and the frame formats and start coding yourself.

An entirely different question is: is it necessary? The answer is in my opinion) not at all. CanZE already digs up module temperatures and cell voltages!! It uses diagnostic messages that are gatewayed through the EVC to the LBS's.

I hop this helps.

gcshave, 2016-03-01:
Thank you for taking the time to answer my questions.

I bought the battery pack for installation in an electric vehicle
conversion. I was hoping to be able to use the LBS (BMS) that came with the
pack without the EVC since I don't have one.

I will need to look at other options.

Thanks again.

Gary

yoh-there, 2016-03-01:
Oh, I see. I think that would be a very hard thing to do, because the BMS has a "digital rights management" system in it. Since in most countries the battery is rental only, they can disable it through the TCU. While it is a very good BMS, you're probably better off building your own. There are fairly decent balancing modules available for cheaps.

closed 2016-03-01 by yoh-there