fesch/CanZE #329: Battery screen is missing info (v1.11)

HarmOtten, 2016-03-05:
1. On this screen the SOH is not showing anymore. On other screens it still does.

2. The scale of the Cell Voltages is max 4 Volt. Since the voltage are above 4 Volt, it would be better to change the scale to max 5 Volt. Or even better, from 2 or 3 to 5 Volt.

fesch, 2016-03-06:
The field was commented out in the Fields class :-(

I've reactivated it.

closed 2016-03-06 by fesch

yoh-there, 2016-03-06:
Oops, I am sorry, my bad. If not done already, please do update the spreadsheet too (Y in column "used in CanZE") so it will be re-exported once we generate those tables again. It is a recipe for introducing regressions ;-)

Pixel91630, 2016-03-09:
Hello,
On the PlayStore, the last version is 1.11, Feb. 15th.
The SOH is still not showing.
How can I get the lats modified version ?
Thanks in advance.

yoh-there, 2016-03-09:
We have no release date planned, but it is fixed in the current source code and will be included in the next release, when it will come out. Further, see http://canze.fisch.lu/downloads/ (about "bleeding edge")

Pixel91630, 2016-03-09:
Thanks for your help.
I found the SOH in an other screen of the application.

MAlfare, 2016-05-16:
Hi
I waited for for  this release so long, but was told today by Jeroen, that it was done already on 11th of April.
Have installed: Version 1.12 // Build: 2016.04.11 @ 07:56.-
But in "Battery" screen, the bottom left field "Cell Voltages (V)" is still totally empty.

Is it only me?
Regards
Manfred