fesch/CanZE #331: Include timestamp in 'Fields' logging

alexdepalex, 2016-03-10:
Version: 1.11 // Build: 2016.02.15 @ 22:56

When selecting 'Fields' from SD card logging, the resulting log file contains keys and values, but not the timestamp which would be convenient when trying to recreate graphs offline. Is there already an option to enable timestamps? If not, is this a feature which could be easily implemented? My suggestion would be te prepend the key with an epoch timestamp. Or use an additional log line with epoch whenever the time changes. The first option would increase the logfile, with about 350KB (12 * 31437) for a 75 minute trip. The second option about 54KB (75 * 60 * 12).

yoh-there, 2016-03-10:
I passed the issue on. Seems like a fair suggestion. I would personally opt for the first option, as it is easier to process in i.e. a spreadsheet, but that's strictly IMHO.

alexdepalex, 2016-03-11:
True regarding the first option. Would be easier and with sdcard sizes nowadays, the increased size wouldn't really be an issue anyway.

yoh-there, 2016-03-11:
@chrismade check if this alerts chrismade. Name did not appear in the address picker after typing the at-sign

chrismade, 2016-03-11:
in the current source code all data-YYYY-MM-DD*.log files start with
5691890732;2016-03-04-21:40:27;3,200;1,000;24.0 ;351.5 V;-2.25 A;0.0 %;18.0 Nm
where first column is a timestamp in millisecs (since 1.1.1970) which is easier for calculating
second column is the same timestamp in human readable format
followed by current data - 

please note there are multiple log files - for multiple purposes:
---
a.) "fields" - mentioned here - belongs to Bob (diag info?)
b.) "debug" - is all debug info 
c.) "data" is the data logging with timestamp

alexdepalex, 2016-03-12:
Are those changes also in the current build? I've enabled the 'data' logging, but that only showed me some static data which didn't change across the entire trip. The 'fields' logging is probably the best candidate to get an extract of all the data, but that isn't timestamped.

Also, it seems like debug logging is directly flushed to sd, which might impact performance or the collection of data in general. I'm totally clueless regarding Java, so I might be wrong regarding this :)

chrismade, 2016-03-12:
There were two major bugs in data logging which have been fixed in Version 1.11 / 2016.02.15 - the version you are using - since then it is working as expected (at least on my end).
One known issue still is the place where the file is stored - from coding it should go to SD card /storage/sdcard1 - at least my version still writes to /storage/sdcard0/CanZE - can you pls check this?

BTW: thanks for using the data logging function

fesch, 2016-03-15:
The change to the "field" logger (that logs only the fields from the actual screen) will be in the next release.

fesch, 2016-04-18:
Release on 2016.04.11

closed 2016-04-18 by fesch