fesch/CanZE #333: Battery screen improvements

yoh-there, 2016-03-17:
Based on suggestion on goingelectric and a long drive by myself yesterday......

- remove SOH graph;
- use real estate to display SOH numerically, and 12V voltage
- change RealSOC field to 7bb.6103.192 (this seems to better represent the bare battery SOC)
- replace battery temperatures with a (color only) heatmap? It's rather unreadable and unclear (to me at least) what the green lines mean.

What do you think?

fesch, 2016-03-17:
Yes, let's see what it looks like ;-)

fesch, 2016-04-19:
change RealSOC field to 7bb.6103.192: done

yoh-there, 2016-07-18:
My suggestion is to close this item (even though I opened it). All values are available through the (tech) charging screen and the heatmap anyway.

closed 2016-12-18 by yoh-there