fesch/CanZE #335: Cell Voltages

MAlfare, 2016-04-14:
... are not displayed any more since long time.
Will they come back some time?
Voltage heatmap looks very nice, but who could get information out of it?
A simple numerical display of:

**number and voltage of the highest cell
number and voltage of the lowest cell
difference voltage of lowest - highest
average voltage**

would give instantly more info

yoh-there, 2016-04-15:
I have to push back here a little. Average voltage is a useless number. It's a "poor man's SOC" for a cell like the Zoe's and serves no purpose at all really. Basically, the same is true for highest and lowest.

What matters is the difference between the mean cell voltage and each and every single cell voltage. And of course it's location, and that i exactly what the heat map gives you, in one blink. 96 bars are, IMHO, lovely eye candy, but totally useless. 

Furthermore, the heat map has the "Nissan bad cell" algorithm implemented, which you will only see if there IS a bad cell. An bad cell will glow up bright red when detected, but there are some conditions to be met before the algorithm is usable, notably the pack being on a low SOC (from memory, too lazy to look it up under 25%)

To be quite blunt (sorry), I understand and appreciate your suggestions #336 about better UI, but I completely fail to see you picking on this one specifically. It is, again IMHO, together with the consumption screen, one of the most intuitive, understandable and "immediate" screens of all, while still giving the user all the nitty gritty details if that's what she likes.

@fesch : I thought the voltage graph was fixed (first question asked). Maybe not released yet? I will close this issue, but if I am mistaken on that part, will you please reopen?

closed 2016-04-15 by yoh-there

MAlfare, 2016-04-15:
@yoh-there 
Average voltage may be useless for you, but it is NOT in general.
It is NOT( Vmax - Vmin)  / 2, it must be sum(V1 .. V96)/96.
And it may be not very informative, if it stands alone, but in conjunction with Vmx, Vmin and Vdiff it can tell me (if not you), if more ore less cells are higher or lower than average.
The vertical bars are a lot better than flying dots, especially with a better (variable) scale! 
Have a look at  [](https://youtu.be/JXGj1bv0giU) minute 2:40.
If you want to know how to do, that, ask Alexandre.

yoh-there, 2016-04-15:
Don't be condecending, it serves no purpose. I know how to calculate an average thank you very much (the color coding uses exactly this), I know how to make bar graphs, and Alexandre is a team member, so please......

Again, the deviation of the mean is indicated by color gradient. Maybe, it is an honest suggestion, no critisism, you are misinterpreting the screen. Anything other than grey is a deviaton of the mean. Redder = higher, blue-er is colder.

We disagree on vertical bars, and anyway, that should work in the graphical representation screen. As I indicated to fesch, if that is non functional in the latest code build, it should be fixed, though I think it IS fixed, though may be not released yet.