fesch/CanZE #336: Some general Thoughts about representing data in CanZE

MAlfare, 2016-04-14:
Hi!

Information should be recognized as quick as possible, with a minimum of attention (especially values of interest during driving).

Small displays cannot permit this, so design should be adjusted for (7 inch ?)tablets, with either landscape or portrait orientation.

It is not a good idea, to separate description and the value belonging to by a lot (10 centimeters) of white space, especially with small fonts. You need a ruler to get the pairs together.

To reduce/avoid the need of changing screens, information should be concentrated. Descriptions could be in small fonts to save space, users will get used to recognize values by the place on the screen after a while, and never read the descriptions.

Lengths of vertical bars are much faster focused/recognized than of horizontal ones (maybe because human eyes are arranged horizontally)

Flying dots (cell voltages) or lines (compartment temperatures) are not intuitive to read.

Voltage heatmap and Compartment heatmap are very nice, but there is (an additional) better way to present this data.
Digital values take a lot of attention to be recognized, specially if there are so many with many of decimal places.
Advantage is their precision.
Fluence ZE Spy from Alexandre Moleiro does many things very well, especially the presentation of the cell voltages! 

As breaking mostly occur during driving, informations from "driving info" and "braking" screens should be displayed on one screen.

"Charging" screen wastes a lot of space an misses very interesting values, available on a different "Technical" screen.

Regards
Manfred

fesch, 2016-04-15:
Thanks for your feedback. We will take a look at it and try to modify things accordingly, but I would like to point out, that not all screens are meant to be used while driving ... in fact, most of them are not.

yoh-there, 2019-01-28:
We will not make further adjustments. In addition to what @fesch wrote, the rationale for the screens and how they are designed is:

- driving: concentrate on real speed, related consumption. remaining energy
- braking: training for minimizing friction braking
Those are basically information-sparse screens designed for driving, but have different purposes

Consumption:
Overall info. Mostly useful at the end of a drive (but I admit you need to keep it open for proper 
population of data)

Charging: consciously dumbed down information. If you need more detail, go to tech. This is exactly as intended.

Temp map of thebattery: I disagree. The colors give an immediate and intuitive feedback. A problematic cell will jump out. The numbers are simply there for detail.

Closing

closed 2019-01-28 by yoh-there