fesch/CanZE #337: Battery kilometers not correct

HarmOtten, 2016-04-16:
CanZE v1.12
I checked the new value [Battery kilometers] on the Detailed Charging screen, and it showed 8565,0.
But my battery has never been replaced, so it should be 74101km. (I double checked with i907)

fesch, 2016-04-18:
On my Zoé the value is the same than the one displayed on the car ....

yoh-there, 2016-04-18:
Rollover on 16 bit number. Noting we can do about that I am afraid. Maybe add in description

HarmOtten, 2016-04-18:
We will find the most significant byte/word of this value somehow, because I know it is there.

yoh-there, 2016-05-14:
Fixed in #339 I cannot test it as I haven't passed 65K kms yet. I will close this issue. Please test your personal pull/clone, or after the next release. Thanks!

closed 2016-05-14 by yoh-there