fesch/CanZE #341: Maybe # 329 should be reopened?

MAlfare, 2016-05-17:
Please read here for details:

https://www.goingelectric.de/forum/renault-zoe-zubehoer/obd2-diagnosegeraet-fuer-renault-zoe-erhaeltlich-t7818.html#p349939

The discussion between Jeroen and me, MAlfare, starting on Do 12. Mai 2016, 01:47 

Regards
Manfred

yoh-there, 2016-05-17:
@fesch  : Version 12, voltage heatmap works. I thought scaling and all was fixed on 12? But I am running the development version of course, so maybe I have been too optimistic.

@MAlfare  please correct if I'm wrong).

MAlfare, 2016-05-17:
Version 1.12 // Build: 2016.04.11 @ 07:56
Voltage heatmap works, scaling is fixed, but no data shown on Battery -> Cellvoltages (V)

fesch, 2016-05-22:
Checking ....

fesch, 2016-05-22:
I just tried multiple times, even with clearing all data, closing the app ... etc and I always get the cell voltages graph on the Battery page filled up nicely ...

MAlfare, 2016-05-22:
Hi fesch!
Additinoal Information:
Android Version of my ZTE-phone: 4.2.2, on my Lenovo-tablet: 4.4.2
Not working on both, with downloaded from Playstore.
Could you pass me your (working) apk for testing?

Regards
Manfred

fesch, 2016-05-22:
Voilà

[2016-05-22-canze.apk.zip](https://github.com/fesch/CanZE/files/276298/2016-05-22-canze.apk.zip)

MAlfare, 2016-05-22:
Thank you!
Works on the tablet, and after some bluetooth problems, app- and system-crashes, on the phone too.
So there must be a difference between the playstore version, and the one. you sent me.

fesch, 2016-05-23:
New version published ...

closed 2016-05-23 by fesch