fesch/CanZE #342: Charging Graphs

MAlfare, 2016-05-30:
Hi!

Some things, that could/should be done on Charging Graphs screen:

First graph (Pilot amp (A) / Charging power (kW)):
 Scaling for Charging Power is from 0 to 40, but ZOE Q210 pull ore than 40, so scaling should be adjusted.

Second one (Max battery charge kW / DC Charging power (kW):
() missing in Max battery charge kW.
more annoying: has no scale on the right side (doesn't really need none), so time lines of the three graphs are misaligned.

Third one Available energy (kWh) / temperature:
Temperature has no scale unit.

[image]

yoh-there, 2016-05-31:
40: The graph represents the DC power. I've never seen it above 40, have you?

Middle and third graph: agrreed!

yoh-there, 2016-05-31:
Fixed in merge #344

closed 2016-05-31 by yoh-there

MAlfare, 2016-07-21:
Hi yoh-there!

You can answer your question by yourself
(40: The graph represents the DC power. I've never seen it above 40, have you?)
by taking a look at the screen shot, where the blue line leaves and enters the diagram area more than one time.

My answer:
Yes I have.
I even was able to make a screen shot ;-)

yoh-there, 2016-07-21:
Hi @MAlfare I will leave this issue closed for this moment. While I agree that IF charging power goes above 40, we could scale the graph up a bit, but it would "uglify" the scale and not add a lot of value. 

More importantly, I am now not entirely convinced that the graphing widget itself is 100% correct under all circumstances and scalings (see the reopened #343 ), so I will wait for that and my next 43 kW charging session so I can cross check with the numerical screen. This could take a while BTW, vacationing and all ;-)

commit 9df1cc95 (2016-05-31): Merge pull request #344 from yoh-there/master

Added 8 bits to HV kilometers (bringing it in line with car kms), attempted fix issue #337, fixed #342