fesch/CanZE #343: Cell Voltages

MAlfare, 2016-05-30:
Hi!

During charging I could see a serious difference between cell voltages in BATT VOLTAGES -Voltage heatmap and BATTERY - Cell Voltages (V).

While voltages of all cells in heatmap are over 4 Volts, all voltages in BATTERY - Cell Voltages are at approximately 3.6 Volts.

Where is the difference coming from?
Are there different sources of data?
Is the difference Ri * Ic (Ri = inner resistance of the cell, Ic = charging current)?

Regards
Manfred

fesch, 2016-05-31:
This is quite strange, as the same CAN messages are being used to obtain both information .... !?

yoh-there, 2016-05-31:
Exactly. @fesch is there any way (and if not, this is a suggestion) to disable pulling the variables from the database and instead only show real time data?

yoh-there, 2016-07-18:
Unable to reproduce. Closing

MAlfare, 2016-07-21:
Hi yoh-there!

I **am** able to reproduce!

You can close whatever you want, it's your party.
But is it good habit, to ignore bugs in that way? 

Have a look at the two screen shots, taken immediately one after the other.

Maybe the values are O.K but the scale is faulty?
[image]
[image]

yoh-there, 2016-07-21:
Double checked against current source. Source (battery.xml) seems right, but I can confirm @MAlfare 's screen print. @fesch could you please have a quick look at the widget code? As I am 99% convinced the heatmap is right (as in: stupid simple, it just prints the value in the proper block)

fesch, 2016-07-28:
Can confirm: got a voltage heatmap with values > 4 but the graph on the battery screen displays them below 4 ....

fesch, 2016-07-28:
Got it! This will be fixed in the next release ...

yoh-there, 2016-07-29:
Great. I would not mind a release. It would fix this, might fix #342 and put the prediction (which was relatively crap in the previous release) in the wild for real user testing ;-) Up to you of course.

fesch, 2016-07-29:
I must admit that I did not yet test the prediction too. Each time I charged the last few days my pad has not right near me ;-)

fesch, 2016-07-29:
But I close this issue now anyway. 
The fix will come with the next release ...

closed 2016-07-29 by fesch