fesch/CanZE #349: Degradation graph and or prediction

yoh-there, 2016-07-04:
Title says it all, per Sandy' suggestion. I think if we decide to implement this, we should be careful with the following two issues:
- collected data should be stored per VIN just in case it is used for another car;
- collected data should not be erased through the clear cache button.

yoh-there, 2016-07-10:
I have given this suggestion a bit more thought. In all fairness, I don't think we can implement this without cloud based storage. Phones line 2-3 years (with probably one or two full factory resets in that time frame); degradation to 75% for a ZOE is expected in the rough order of 200.000 kilometers, which is much more than the lifespan of the device.

Given the reluctance to implement and also maintain ANYTHING with remote data, partly because of Luxembourg and German legislation, I think this might be something we need to let go.

@fesch what do you think?

fesch, 2016-07-10:
You are right: As soon as we make our app "connected", we will get a rather large tail of constraints to handle to make this "the right way".

Although any user must upload his data 100% willingly I think that the stored data itself will need some extra protection we could hardly guarantee. And then, as you already mentioned, we would need to ask to be allowed to handle such a database ... to be 100% legally ...

yoh-there, 2016-07-10:
Would storing in file in a users PERSONAL google drive be OK?

fesch, 2016-07-11:
Probably, but I think putting a file there from within the app automatically will not be trivial. To be tested ... I've never done that ...

yoh-there, 2016-07-11:
I skimmed this https://developers.google.com/drive/android/ Not VERY hard it seems, and I am slightly unsure about the keys required and if OAuth can be automated to ones own identity. Or I should maybe say: I am 100% that can be done, but unsure about the complexities :-)

yoh-there, 2016-12-19:
I am now leaning to a similar, but alternative approach, What about using a well established IoT service like thingsspeak? Strictly personal, I don't want to have to do with ANY storage responsibility myself.

Mildly related: next version of CanZE needs to have Internet access in the manifest as it needs to be able to access a Wemos Gateway or an emulator. I have already drafted a note on that change for the users on the CanZE website.

yoh-there, 2017-01-16:
I have made a draft note on MQTT on the CanZE website. Don't get me wrong, this is NOT a given. I am OK with not publishing that all. But I do like the idea and for sure I am going to implement it on the WeMos

fesch, 2017-01-16:
Even without heaving read it right now,
I think "why not". Maybe someone gets inspired ;)

Bob

yoh-there, 2017-01-16:
In that case I'll just publish it and we'll see. If there is lukewarm response, we'll only do it for ourselves if and when we want it :-)

yoh-there, 2017-01-30:
I am closing this issue. MQTT is on the back burner as far as I am concerned, in the sense that I will implement it on the WiFi gateway first. As we will not implement a degradation function in CanZE itself for the reasons stated above, there is no need to keep this issue open.

I do expect keen interest from a few number-keen users to push for MQTT and implement the most interesting graphs and stuff on degradation cloud based later on.

closed 2017-01-30 by yoh-there