fesch/CanZE #353: DATA log to sdcard mostly not working

joergze, 2016-10-02:
Version 1.15 Build 2016.08.10
Dongle ELM327 Maxiscan OBDII
Android Device: Motorola Moto G2 with Android 6.0 or HTC Desire SV with Android 4.0.4

"Log to sdcard1" is active, but there is mostly written only an empty file. But not always. Sometimes there are some useful looking data in it, but file is never bigger then 150 KB, recording seems to stop after some seconds.

Showing the Information in the Consumption or Driving windows works.

The field and debug log seems to work

Loc17, 2016-11-17:
Hi Joerg,

I seems to have the same problem with Log to sdcard, it never works for me, all the datas are logged to device under CanZE folder. I also cannot read the log files, they are many columns and values, how can you define each value?

fesch, 2016-11-18:
What do you mean by "cannot read"?
Is that something like "I don't know how to interpret the content of the file ..." or
more like "The file is not present on the device ..." or 
rather "The file is there but I get an I/O exception when accessing from it ..." of
finally "The file is present but I do not have the correct permission to read it ...".

Loc17, 2016-11-18:
Sorry for making unclear statement xD
I mean I don't really know how to interpret the content. I know there are values from the fields log, but also there are some very long long numbers and so on. Do you have an example of a log file in which you define all the values from the file and then able to make a table or something to sort them out offline?
Another question, I asked about this on CanZE site. The option "fields" log, it will log the current visible screen, but when I turn to another screen (safe driving mode off), it will continue to write in the file, which is already written from another screen, and therefore makes it harder to separate 2 log contents from the 2 screens right? How can I stop 1 file (from 1 field) and continue to log to another different new file?
Thanks

fesch, 2016-11-18:
Okay :-) ...  so I do not have to look into to code to detect some problem. Yes, I will provide mor information about the logged format.

If you switch from screen-1 to screen-2, fields from screen-1 will no longer be logged.

A more detailed logging, with selectable fields has been planned but not yet implemented due to more severe problems with the connection that had to be resolved prior to logging.

fesch, 2016-11-18:
The first line of the log file reads: time,SID,value
From that you should be able to derive that the first column contains the "timestamp", the second one the ID of the read field an the third one the associated read value. Columns are separated by a simple comma.

Loc17, 2016-11-22:
Hi Bob, I tried to read again, but I still can't interpret the file.

For example the file goes like this 
2016-10-31-09-28-30,5d7.0,399
2016-10-31-09-28-30,1f8.16,2700
2016-10-31-09-28-30,1f8.28,4094
2016-10-31-09-28-30,1f8.40,21
2016-10-31-09-28-30,427.40,170
2016-10-31-09-28-30,427.49,141

I think the time is 5d7, 1f8, 427. There are more in the log file like 7ec, 7bb. How can I interpret that time column?

There are also many SID of values, which yea I also can't understand which is which, although they are from the same field. And their values also don't show quite clear which type of value are they. For example

2016-10-31-09-28-34,7ec.623204.24,32756
2016-10-31-09-28-34,7bb.6142.16,3882
2016-10-31-09-28-34,7bb.6142.32,3881
2016-10-31-09-28-34,7bb.6142.48,3882
2016-10-31-09-28-34,7bb.6142.64,3882
2016-10-31-09-28-35,7bb.6142.80,3881
2016-10-31-09-28-35,7bb.6142.96,3881
2016-10-31-09-28-35,7bb.6142.112,3882

Can you please explain me those? I also have few more questions. How do you read/work with the log files? Is it the same with my files? Is there anyway to sort them up or make them clearer to read?

When will the new features of logging will be implemented? 

And if I choose "log to SDcard", which fields will be logged exactly?

yoh-there, 2016-11-22:
Let me barge in here.....

Loc, please PLEASE do read the answers (I am not trying to be rude), let me quote....

"The first line of the log file reads: time,SID,value"

Your example

2016-10-31-09-28-30,5d7.0,399

How can you state that 5d7 is the time? As Bob said, 2016-10-31-09-28-30 is the time.  09:28. The SID is 5d7.0, The value is 399. I repeat after Bob: It is a **comma** separated file.

If you want to know what the SID means, well, you have to dig into the source code (Fields.java) from memory. It is all open source and available here on github.

All your other questions were answered by Bob earlier. There are no plans to make "easier readable log files".

Loc17, 2016-11-22:
[image]
 Sorry for the misunderstanding. I also don't really think it's time, but as my reader shows, it's in line with time and I don't really have experience before. Thanks for your answer, I'll try to look for them

yoh-there, 2016-11-22:
Okidoki.

BTW, 5d7.0  is speed.

Loc17, 2016-11-22:
yea thanks for your help, I'm sorting out the SID now. I really don't know much about Java and Open Source before but this is still okay for me. I found most of the SID in CanZE/app/src/main/java/lu/fisch/canze/activities/, others in CanZE/app/src/main/java/lu/fisch/canze/classes/DataLogger.java
But there are still some SIDs unknown for me, maybe I have to dig in more. Can I ask here if I have some more questions or should I create a new topic?

Loc17, 2016-11-22:
Hi yoh there,

I search for Fields.java but there is no result for that. The datalogger.java is somehow not the one that prints the fields logs out. I read the code, it's totally different. Can you please show me how to find it?

fesch, 2016-11-22:
find * | grep Fields.java
app/src/main/java/lu/fisch/canze/actors/Fields.java

Loc17, 2016-11-22:
Hi Bob, hi yoh,

I read the code file, as long as I understand, this file will create the field.log
https://github.com/fesch/CanZE/blob/ec2bd195a5af97558d185cc7aae4db26d6ea09e8/app/src/main/java/lu/fisch/canze/classes/FieldLogger.java

Which lead to 
https://github.com/fesch/CanZE/blob/ec2bd195a5af97558d185cc7aae4db26d6ea09e8/app/src/main/java/lu/fisch/canze/actors/Fields.java
to write down the timestamp, SID, value

but one thing I can't understand. The value writen in the file for realSpeed is for example 3629, 3793 etc.
Is that 36.29 km/h? How can I find in the code which value will have which unit in which format except from this? For example Torque value 3475, is that 347.5 Nm or 35.75 Nm? 3575 Nm is not possible right?

[image]

Does this has anything to do with these lines? I don't quite understand the algorithm

[image]

Also another question, there are some differences between the SID in the codes and in the log files, like 
1f8.28 for ElecBrakeWheelsTorqueApplied 
but in the file there are 1f8.16, 1f8.40. 
It is just an example, there are many more different numbers from a SID.

Can you help me to interpret this? I'm quite bad at programming and it just happens that I need to dig into this field for the project.

Thank you so much

fesch, 2016-11-22:
if 1f8.28 is ElecBrakeWheelsTorqueApplied, 1f8.16 and 1f8.40 are obviously something else.

The lines of codes interpret a signed integer as what it is.

yoh-there, 2016-11-22:
Stay away from the field logger. You **ONLY** want to look at
https://github.com/fesch/CanZE/blob/master/app/src/main/java/lu/fisch/canze/actors/Fields.java

Really, believe me :-)

Now look at Line 296

String fieldDef = // ID, startBit, endBit, resolution, offset, decimals, unit, requestID, responseID, options is HEX

Look at line 812
**+"0x1f8,28,39,-1,4096,0,Nm,,,ff\n" // ElecBrakeWheelsTorqueApplied**

ID -> 1f8
startbit -> 28
endbit -> 39
resolution  -> -1 
offset -> 4096
decimals -> 0
unit -> Nm

1f8,16 is ONE line above
**+"0x1f8,16,27,1,4096,0,Nm,,,ff\n" // TotalPotentialResistiveWheelsTorque**

Check that file, and only that file, and you are golden.

Did you contact mr Baert?

Loc17, 2016-11-22:
Hi, you are Jeroen!

Yes I have contacted Mr. Baert. I think he will write back in few days.

Speaking about these lines, can I asked you some more? It is really interesting. From line 813

[image]

I think in the app it doesn't show Engine RPM. But this one, is it the Electro engine RPM which is correspondent in RPM * Torque = Power? If it's so then although it's not shown in the App, it is still logged out because I found the value of this SID in the log file. Is my statement correct?

And so I was thinking, is there the Torque from the Engine, so I don't have to calculate the Wheel Torque to Engine Torque with the fixed transmission gear 9,32. But I found alot of Torque in the code and I found some Torques like:
Mean Effective Torque
Coasting Torque
Total potential resistive wheel Torque
ASR dynamic torque request
ASR static torque request
MSR Torque
But I think those are not Engine Torque. Still there are from line 1191, 1192 and 582, 583 

[image]

What are those Torque actually?

Everything else is well explained with your answer. Like speed with 2 decimals.
Thank you very much guys.

yoh-there, 2016-11-22:
Your statement is correct. We do query motor RPM to calculate frictionPower. That is why it shows up in the log. If you want to calculate power from it, of course you must be very sure to use motor torque, not wheel torque. But I hope you are not trying to calculate power that way..........

The other torques you have to figure out yourself mostly, we just know their names and sometimes we know what they do. i.e. coasting torque is the motor braking torque when just letting go of all pedals. Total potential resistive wheel torque is EXACTLY what it says! ;-)

Loc17, 2016-11-22:
Yea I won't calculate the Power that way, I want to use Power and Motor RPM to calculate back to the motor torque. Is the Power we use to calculate here the Value dcPwr, the DC Power?

Sorry to make it unclear but I didn't ask you to explain all those values :P I also think they are not Motor Torque, but one question, the torque requested and applied from line 1191 1192 and 582 583 are also not Motor Torque?

yoh-there, 2016-11-22:
I think most are motor torques, but I am not sure. Some logging will show it.

The "driving" (forward) motor torque is.

+"0x186,16,27,0.5,800,1,Nm,,,ff\n" // MeanEffectiveTorque

No need to reverse-calculate that.

Loc17, 2016-11-27:
Thanks, I also wonder about this
I can't seem to find Motor current in the logs. There is Motor Voltage with the description in the Source code
[image]
 so I wonder if I can find the Motor Power to calculate back to the Motor current. But it seems like there is also no Motor Leistung log. And I found this, do you know what it is?

[image]

As far as I understand, the DC controller take the power from the battery and deliver it to the DC motor, so is the dcPwr = the Motor power? Can I take the value of dcPwr to Motor Power?

Thank you

yoh-there, 2016-11-27:
The motor is a three phase externally excited AC motor, not DC. The inverter (PEB) takes the 400 volt from the battery and converts it to 3 phase AC. When we look at the power levels though, we tend to look at what goes into the PEB. It's not really useful to look at the motor without the PEB anyway.

The PEB is controlled mostly through the EVC, which is the box you're talking to with those 7ec commands. I think 7ec.623204.24 is exactly what you are looking for.

Loc17, 2016-11-30:
hmm, so you are saying the 7ec.623204.24 is the Motor Current, and therefore 7ec.623203.24 should be the Motor Voltage?
So what about the line above that say 7ec.622004.24 is Motor Voltage?

I'm also looking for Motor Temperature. There are some suspections but I can't be sure about it, do you have any idea what temperature is this?
[image]

and What is the different between Temperature and Interior Temperature below?

[image]

[image]

Thanks alot Jeroen

yoh-there, 2016-11-30:
> hmm, so you are saying the 7ec.623204.24 is the Motor Current, and therefore 7ec.623203.24 should be the Motor Voltage?

I believe it is what it says: Traction battery voltage and current, so that includes other consumers. In real life, those consumers take no more than 5% of total power, unless in severe cold or heat (A/C)

> So what about the line above that say 7ec.622004.24 is Motor Voltage?

I don't know

> I'm also looking for Motor Temperature. There are some suspections but I can't be sure about it, do you have any idea what temperature is this?

No, but a hint may be by checking in the Frames class what ECU transmits frame 0x534. If it is the motor controller, a good guess would be the motor or the PEB. If it is the A/C, well, no so much. If I head to guess this is ambient temperature, but you would have to test that.

> and What is the different between Temperature and Interior Temperature below?

I don't know

Loc17, 2016-11-30:
Hmmm, I don't know for sure but from these lines, I think 7ec.623204.24 is the HV_Battery Current and 7ec.623203.24 is HV_Battery Voltage, not from Motor.

[image]

Also from this line, I don't understand why Current have unit %?

[image]

Thanks for your answer above too

yoh-there, 2016-11-30:
> Hmmm, I don't know for sure but from these lines, I think 7ec.623204.24 is the HV_Battery Current and 7ec.623203.24 is HV_Battery Voltage, not from Motor.
Which is EXACTLY what I said

> Also from this line, I don't understand why Current have unit %?
That is most probably an error. Thanks for pointing that out.

Loc17, 2016-11-30:
Hmm, I still not quite sure about this although I try to read again and again what you said. I guess the Traction battery value OR so-called HV-Battery value are what go into the PEB right?  And from these lines, I guess the dcPwr is then the Consumption state in W, which shown in the Consumption tab in on-app screen as Econometer right?
[image]

But then there is no Value taken from the Motor itself? Like the Motor power? What I really look for is the Motor Power ...

Loc17, 2016-12-05:
Hi again :D,

Can i ask the different between State of Charge 654.25 and real State of Charge 7bb.6103.192
At the same timestamp (second 33 and second 34) the real SOC shows 81,95% while the SOC shows 86%

yoh-there, 2016-12-14:
I am closing this issue as I don't think that at the moment there is anything wrong with the logging code in CanZE, though admitted it is somewhat limited. Anyone feel free to reopen.

Loc: I cannot explain things clear than I have and we have no formal definition of SOC vs. RealSOC vs. UserSOC etc etc. I you find out more, we'd love to hear it.

In an upcoming version of CanZE there will be more diagnostic definitions, but I guess that would be too late for your project.

closed 2016-12-14 by yoh-there