fesch/CanZE #357: R240 not showing data

zmeglic, 2016-11-03:
v1.17
R240 Zoe
Konwei OBD

When Q210 is selected everything works fine, but with R240, most of data is missing.

[image]

yoh-there, 2016-11-03:
Thank you.

@fesch I will pick this one up tonight :-(

fesch, 2016-11-03:
@yoh-there : OK, probably the fields are wrongly assigned ...

zmeglic, 2016-11-03:
Also check other pages. Almost all have issues with R240.

yoh-there, 2016-11-03:
More in the fields class. The non-ff options :-(

I think I have it all covered, but a second pair of eyes cannot hurt, so I
will do a merge request, but won't merge it.

The only thing that didn't come up properly was the Battery activity, but
for the life of me, I can't see why. As far as I can see all the fields
used are properly configured. The only thing I didn't do was reset the
database.

Oh and I didn't update the changelog, not until you've had at least a quick
test :-)

Sorry for the mess Bob.

Jeroen

closed 2016-12-14 by yoh-there