fesch/CanZE #359: R210 - not showing any weel pressure

Dom78, 2016-12-13:
I’ve got a Maxiscan KW902 and tried it with my Samsung Galaxy S5. It seams not to working fine. I can get some informations like the SOC or the Battery Heatmap but it doesn’t show me the weel pressure. On this and other sites it shows very fast some error or info messages. I’ve tested it at a R210 and Q240 with the same results. Any one a idea what goes wrong here? Thanks for any ideas.

yoh-there, 2016-12-13:
I do not have a definite answer, but I can tell you tire pressure monitoring is a complicated subject.

1) Tire pressure monitoring has become mandatory per (from memory) 2014-11. I know delivery of ZOEs with a TPMS receiver (meaning, the system with the transmitter valves) started only very, very close to that deadline. There is a distinct possibility that your car doesn't have the required hardware. It would be a very good idea in my opinion to check with the dealer, or, if you are brave, let a good bit of air out of one wheel and drive for bit. Remember those valves do not work out of the box, they need to be "learned" by the car using the dealer CLIP tool.

2) Next, these valves do not transmit continuously. They only report themselves in once in a while (in the order of once a minute) and they shut down themselves when you are not driving so save their tiny battery. So even if you do have the hardware and everything is A-OK, do NOT expected any results after driving for at least a minute. Testing on the driveway is simply impossible.

3) Then, just the other day I learned that Renault has abandoned the TMPS valve system recently and replaced it with careful monitoring of the rotation of the wheels to detect an unbalance in circumference. We haen't had our hands on a system like that and we don't know what messages it sends on the bus. We don't even know if it is capable of guessing the pressure at all, as opposed to just a "something is not OK".

Reading your message it seems like 3) is out of the question (car not recent enough), 1) and 2) a possibility that you unfortunately have to check for yourself and we'd love to hear from you.

At this moment there is not a whole lot we can do. As long as that debug line is changing CanZE does query the modules, it seems that the car simply does not report anything useful.

yoh-there, 2017-01-07:
Closing due to lack of follow up. Feel free to re-open when there is more info.

closed 2017-01-07 by yoh-there