fesch/CanZE #371: Battery page on canZE not populating

zoezen, 2016-12-22:
I have been using my V1 .17 Konnwei OBD 11,  00:1D:A5:16:32:8B , build 1979 11.30 @ 00.00  for some time now with no problems, but for several weeks now the BATTERY page on my galaxy S2 phone does not show any readings (including State of Health).
I use canZE mainly to monitor charge rates etc which works well and also shows the four sets of data at the bottom of the screen including SOC and SOH.
Other pages relating to the battery voltages and modules work.
How can this be fixed?

yoh-there, 2016-12-22:
Version 1.18 has been out for a while, and I am pretty sure your issue is fixed in it. Can you please report back so we can decide to either close or escalate this issue?

zoezen, 2016-12-24:
Sorry for delay, bit busy just now. My phone is low on memory and hadn't done the update which was asking for 100mb space? I checked with my tablet which had updated and all is well.
Happy Christmas to all and looking forward to another exciting year of development for Zoe.
Many thanks to @yoh-there and the crew on canZE.

yoh-there, 2016-12-24:
Thanks for reporting back. Closing issue. Happy Xmas :-)

closed 2016-12-24 by yoh-there