fesch/CanZE #378: Request for multi-language

yoh-there, 2016-12-26:
Title says it all. User Twizyflu has offered to do the German translation, but of course we have not been very careful about separating out Strings as resources. This would be needed before any translation work makes sense.

Comments please if this is required / worth the effort.

fesch, 2017-01-07:
- [ ] Dutch
- [ ] English
- [ ] French
- [ ] German
- [ ] Luxembourgish ;-)

yoh-there, 2017-01-07:
Working on it. Overall "delaminating" should be done today, and Dutch as ProofOfConcept. Please do NOT start, changes are massive overall and in progress

fesch, 2017-01-07:
We will then need a new preference in the Settings too ...

yoh-there, 2017-01-07:
Not really? Android resources follows system setting, any string taken from the language specific XML, if not found, fallback to default?

fesch, 2017-01-07:
Hmmm ... but wouldn't it be nice to allow the user to switch to something else than the system default?

yoh-there, 2017-01-16:
I will leave that to you ;-) In the mean time. Nearly all is "resourced out", including a Dutch language file. French and German versions I have send to volunteer native speakers.

yoh-there, 2017-01-30:
Am I right when I say
- Luxembourgish was not really serous? ;-)
- That hard setting a locale is not really a requirement?
- That we can drop in languages as we get files and are not particularly waiting for language x?
If the answer to all the above is "Yes", I think we can close this issue.

I will add a post to the blog asking for volunteers to make language files.

fesch, 2017-01-30:
3x Yes ---> closing ;)

closed 2017-01-30 by fesch