fesch/CanZE #393: Dongle validation

Francois64, 2017-01-05:


Francois64, 2017-01-05:
Hi
I'd like to test Canze, downloaded Canze yesterday to be used with the dongle i already use with other cars.
It is labeled OBDclick, probably from China.
Canze can connect to Bluetooth dongle but can't access to ZOE with an error message to fast to be recorded.
i tried to diagnose the dongle running "ELM327ident.apk" from googleplay and got this result (see below).
Is it possible to say if the dongle is ok for Canze or not ?
Regards.

Canze version 1.1.8 on SONY Xperia Z5c - Android 6.0.1
ZOE R240 03/2016

Scan date=2017-01-04 22:33:59
Device name=OBDII
Device mac=00:1D:A5:68:98:8B
Device Version (Declared)=ELM327 v2.1

Command	Version	Result
AT@1	1.0	OK
ATSPA6	1.0	OK
ATTPA6	1.0	OK
ATTP6	1.0	OK
ATSP00	1.3	KO
ATSP6	1.0	OK
ATAL	1.0	OK
ATNL	1.0	KO
ATAMC	2.0	KO
ATAMT20	2.0	KO
ATRA70	1.3	KO
ATSR70	1.2	KO
ATAR	1.2	OK
ATAT0	1.2	OK
ATAT2	1.2	OK
ATAT1	1.2	OK
ATBD	1.0	KO
ATBRT0F	1.2	OK
ATCAF0	1.0	OK
ATCAF1	1.0	OK
ATCEA04	1.4	KO
ATCEA	1.4	KO
ATCF00000111	1.0	KO
ATCF111	1.0	OK
ATCFC0	1.0	OK
ATCFC1	1.0	OK
ATCM00000111	1.0	KO
ATCM111	1.0	OK
ATCP18	1.0	KO
ATCRA7C0	1.3	OK
ATCRA000007C0	1.3	OK
ATCRA	1.4b	KO
ATCS	1.0	KO
ATCSM0	1.4b	OK
ATCSM1	1.4b	OK
ATCTM5	2.1	KO
ATCTM1	2.1	KO
ATCV0000	1.4	KO
ATD1	1.3	OK
ATD0	1.3	OK
ATDP	1.0	OK
ATDPN	1.0	OK
ATE0	1.0	OK
ATE1	1.0	OK
ATFCSD0430FF00	1.1	KO
ATFCSH000007B0	1.1	KO
ATFCSH7B0	1.1	KO
ATFCSM0	1.1	KO
ATFE	1.3a	KO
ATH0	1.0	OK
ATH1	1.0	OK
ATI	1.0	OK
ATIB10	1.0	OK
ATIB48	1.4	OK
ATIB96	1.0	OK
ATIFRS	1.2	OK
ATIFRH	1.2	OK
ATIFR0	1.2	OK
ATIFR2	1.2	OK
ATIFR1	1.2	OK
ATIGN	1.4	OK
ATIIA7A	1.2	OK
ATJS	1.3	KO
ATJE	1.3	KO
ATJHF0	1.4b	KO
ATJHF1	1.4b	KO
ATJTM5	1.4b	KO
ATJTM1	1.4b	KO
ATKW	1.3	KO
ATKW0	1.2	KO
ATKW1	1.2	KO
ATL0	1.0	OK
ATL1	1.0	OK
ATM1	1.0	OK
ATM0	1.0	OK
ATPPFFON	1.1	KO
ATPPFFOFF	1.1	KO
ATR0	1.0	KO
ATR1	1.0	KO
ATRV	1.0	OK
ATS0	1.3	OK
ATS1	1.3	OK
ATSDFF	1.4	KO
ATRD	1.4	KO
ATSH000007E0	2.0	KO
ATSHA848F1	1.0	OK
ATSH7B0	1.0	OK
ATST32	1.0	OK
ATSWFF	1.0	KO
ATTAF9	1.4	KO
ATV1	1.3	KO
ATV0	1.3	KO
ATWMC133F13E	1.0	KO
ATPC	1.0	OK
ATWS	1.0	OK
ATD	1.0	OK
ATZ	1.0	OK

fesch, 2017-01-05:
Im sorry, the version v2.1 is not supported as it does not allow do make all necessary queries. This can be read in big letters on the corresponding page.

closed 2017-01-05 by fesch

Francois64, 2017-01-07:
I’m sorry too.

I thought that 2.1 version was better than 1.3 or 1.5 and able to do the same as 1.3 & 1.5 plus the gap to 2.1 .

 

François Malaterre

yoh-there, 2017-01-07:
It is true! For the REAL ELM327 chips. However, all the stuff on ebay are knockoffs and they do not run real ELM software and simply claim to be 2.1. This is the source of all this confusion. Get a KONNWEI 902

Francois64, 2017-01-07:
Ok thanks for your advise.

Ihad a look to Ebay but the link i was given by someone fom ZOE forum says « product unavailable for the moment ».

So ihave to wait a little bit.

Next week i’m gona test a dongle from another member of the forum i will met.

Have a good day.

François Malaterre

yoh-there, 2017-01-07:
http://www.ebay.com.au/sch/i.html?_nkw=KW902

>30 results, <15 euro