fesch/CanZE #405: EVC dtc reading

FredLeudon, 2017-01-20:
Hi,

I have a problen when I try to read DTCs from EVC.

Below the result of ios CanZE logs (first, trying to read DTC from EVC and second successful read TCU DTCs)

I get a NO DATA or CAN ERROR each time.

20/01/2017	12:05:51,00	     Device	RealField	Value[] ->  
20/01/2017	12:06:28,46	     CanZE	 DTCActivity
20/01/2017	12:06:28,91	     DTCActivity	Constructeur()	Stopping poller thread
20/01/2017	12:06:29,17	     Device	 Annulation tache parallèle
20/01/2017	12:07:09,22	     ELM327 : initDevice (1)
20/01/2017	12:07:10,09	     Device	ELM[ELM327 v1.5]
20/01/2017	12:07:11,29	     cDevice	requestFrame[7ec.5902ff]
20/01/2017	12:07:11,30	     ELM327	requestIsoTpFrame : [7e4]
20/01/2017	12:07:11,76	     socketManager	frame[7ec]	sendCommand command[031902ff]	timeout(ms)[750]	addReturn[oui]	stopCar[>]
		hexdata[NO DATA<cr><cr>>]
		queryTime[+000000000000023]
		timeout[non]
		stop[oui]
20/01/2017	12:07:11,76	     cDevice	requestIsoTpFrame() -> QueryTime[00:00:465]
20/01/2017	12:07:11,76	     Device: request for 7ec.5902ff returned error -E-data empty
20/01/2017	12:07:25,98	     Frames	Ecu does not exist:LINSCH
20/01/2017	12:07:26,76	     ELM327 : initDevice (1)
20/01/2017	12:07:27,76	     Device	ELM[ELM327 v1.5]
20/01/2017	12:07:29,10	     cDevice	requestFrame[7da.5902ff]
20/01/2017	12:07:29,11	     ELM327	requestIsoTpFrame : [7ca]
20/01/2017	12:07:30,03	     socketManager	frame[7da]	sendCommand command[031902ff]	timeout(ms)[750]	addReturn[oui]	stopCar[>]
		hexdata[10AB590239AEF016<cr>2110AEF04910AEF0<cr>224A10AEF04B10AE<cr>23302910AE222910<cr>24AE232910AE2462<cr>2510AE124910AE10<cr>2649109324111093<cr>27241310AE211210<cr>28933B1210933B11<cr>2910930D1210930D<cr>2A1110930D1310AE<cr>2B311210AE311110<cr>2CAE311310930512<cr>2D10930511109305<cr>2E13109328121093<cr>2F28111093281310<cr>20AE839410AE1306<cr>2110AE322910AE34<cr>222910AE917310AE<cr>23927310AE339610<cr>24AE81491093D98F<cr>2528930212109302<cr>26111093021310AE<cr>27064910AE044910<cr>28AE01F028505050<cr><cr>>]
		queryTime[+000000000000065]
		timeout[non]
		stop[oui]
20/01/2017	12:07:30,05	     cDevice	requestIsoTpFrame() -> QueryTime[00:00:937]
20/01/2017	12:07:33,34	     DTCActivity	Destructeur()	Re-Starting poller thread
20/01/2017	12:07:37,17	     Frames	Ecu does not exist:LINSCH
20/01/2017	12:07:37,75	     Device	Lancement de la tâche parallèle

yoh-there, 2017-01-20:
> hexdata[10AB590239AEF0162110AEF04910AEF0224A10AEF04B10AE23302910AE22
> 291024AE232910AE24622510AE124910AE10264910932411109327241310
> AE21121028933B1210933B112910930D1210930D2A1110930D1310AE2B31
> 1210AE3111102CAE3113109305122D109305111093052E13109328121093
> 2F2811109328131020AE839410AE13062110AE322910AE34222910AE9173
> 10AE23927310AE33961024AE81491093D98F252893021210930226111093
> 021310AE27064910AE04491028AE01F028505050>]

But the ECU LINSCH really does not exist!

The easiest ECU to query is the CLUSTER. If you have CLUSTER running, your
code base is fine.

I hope I didn't misunderstood the problem.

FredLeudon, 2017-01-30:
Code base is fine, I can retrieve CLUSTER DTC's

When I ask for EVC DTC's, ELM returns NO DATA

Any idea ?

yoh-there, 2017-01-30:
Ah OK, got it. 

I think the problem is that you compute the code to send (1902ff), instead of taking it from the definitions (19023b).

FredLeudon, 2017-01-30:
Ok, i see.

How does it work on android CanZE project.

RequestIsoTPFrame computes requestID and will also return 1902ff for 5902ff responseID ?

so requesting EVC DTC with android CanZE will return same error, right ?

I have to add requestID in field class and replace the frame.getrequestId by field.getrequestID in requestIsoTpFrame ?

yoh-there, 2017-01-30:
In the Android code the field is found by responseID (5902ff). There is no specific reason why, but that is littered all over the code base ;-)

When sending an IsoTp command, we find the field, then get (instead of compute) the requestID (as initialized by the Fields), so in this case (19023b). So yes, I think what you propose is the safe way to do it, as there will be more ECU classes added soon.

FredLeudon, 2017-01-30:
but, if I look at the code,

ELM327.RequestISOTPFrame()

 // 022104           ISO-TP single frame - length 2 - payload 2104, which means PID 21 (??), id 04 (see first tab).
        String elmCommand = "0" + (frame.getRequestId().length() / 2) + frame.getRequestId();
        //MainActivity.debug("R: "+request+" - C: "+pre+field.getToId());

and Frame.getRequestId()

  public String getRequestId () {
        if (responseId.compareTo("") == 0) return ("");
        char[] tmpChars = responseId.toCharArray();
        tmpChars[0] -= 0x04;
        return String.valueOf(tmpChars);
    }

The code changed, now Frames are requested instead of Fields ....

yoh-there, 2017-01-30:
Class ELM327, it used getRequestId(), which is initialize using load()

>         String elmCommand = "0" + (frame.getRequestId().length() / 2) + frame.getRequestId();

On the application level, fields are requested. This results in getting the corresponding frame and triggering all the embedded fields with new values. It is slightly more complicated for the diagnostic frames, as the same frame can be used for many different queries. However (example: module temperatures), she same is still true: the application asks for one field, the entire frame is requested and all embedded fields (some just one, but some contain many, many fields) are triggered. The only difference is that at initialisation of the Frame instances, subframes are created for each diagnostic command.

At the end of the day, a device driver only knows what a frame is. It has no clue about fields at all.

FredLeudon, 2017-01-30:
Sorry, but I read the whole code and i'm stil don't understand. 

I will have a look tomorrow if i can.

Thanks for your help.

yoh-there, 2017-02-10:
After some private emails back and forth we have established it has nothing to do with the code, but some misinterpretation on my end. I had assumed the "give me all DTC" command was the same for all ECUs and this is not the case. I now have reasonable documentation on the proper command per ECU but I want to be reasonably sure the response format is the same too.

Assuming (for the moment) that this is true, only an update to the Fields class initialization strings is required.

FredLeudon, 2017-02-10:
Cool, great news !

If possible, would you agree to skate tout documentation ?

Frédéric RICHARD

yoh-there, 2017-02-10:
No problem.

yoh-there, 2017-02-10:
Example for the EVC

Command: 1902AF

Response on my car:
5902ff
04116828
1525f368
06019268

(Line breaks by me). Meaning
0411 - 68 - 28 -> Battery Charger Bloc in Default (from CAN), event information, DTCStatus.testNotCompletedThisMonitoringCycle + DTCStatus.confirmedDTC

1525 - f3- 68 -> Cruise Control/Speed Limiter Deactivation,  Multiple Failure, DTCStatus.testNotCompletedThisMonitoringCycle + DTCStatus.testFailedSinceLastClear + DTCStatus.confirmedDTC

0601 - 92- 68 -> Electric Motor Prestation, performance or incorrect operation,  DTCStatus.testNotCompletedThisMonitoringCycle + DTCStatus.testFailedSinceLastClear + DTCStatus.confirmedDTC

After a quick change in the code (needs fixing up, so uncommitted)
[image]

yoh-there, 2017-02-11:
@fesch @FredLeudon Given that the command to inquire the DTCs is not the same for every ECU, and that it is even possible for an ECU to have more than one command (I assume for the moment this is due to possible sub-systems in an ECU, an example is the BCB / JB2), I will make a change to the Ecu / Ecus classes to include a list of commands to inquire it's DTCs, and of course change the DtcActivity accordingly.

FredLeudon, 2017-02-11:
Ok, let me know on witch branch You do that et n github please.

I will report changes in ios code. 

Thanks 

Frédéric RICHARD

yoh-there, 2017-02-12:
Done in "development". Also added TCU.

FredLeudon, 2017-03-03:
Sorry to reopen but I still have trouble to retrieve DTCs

When trying to get EVC DTCs there is what I do :

- get GetDTCField -> 7ec.5902af.0
- Request Frame -> requestIsoTpFrame(7e4)
- Send command : atsh7e4 -> Ok
- Send command : atfcsh7e4 -> Ok
- Send command : 031902af - > CAN ERROR or NO DATA

Please, say if I'm right with this sequence ?

FredLeudon, 2017-03-03:
other question : 

in EcuDiagEVC class, at the end of load function, I think there is an error

Frames.getInstance().load("7EC,0,0,EVC\n");
Fields.getinstance().load(fieldDef1);
**Fields.getInstance().loadMore(fieldDef1);** _isn't it wrong? fieldDed1 is already loaded ?_
Fields.getInstance().loadMore(fieldDef2);
Dtcs.getInstance().load(dtcDef, testDef);

FredLeudon, 2017-03-03:
.

yoh-there, 2017-03-03:
> Fields.getInstance().loadMore(fieldDef1); isn't it wrong? fieldDed1 is already loaded ?
That is a dumb mistake! Thanks

> Please, say if I'm right with this sequence ?
It SHOULD be right. I will try again this weekend. Tell me, do you have a Q210 or an R240

FredLeudon, 2017-03-03:
i have a Q210 - 03/2013

yoh-there, 2017-03-06:
I can confirm issuing a "get DTCs"command (1902AF) not only gives no
answer, but in my case knocks off the dongle too and I need to reinitialize
it. This was through my gateway though, not sure if that secondary problem
is in the gateway. I can also confirm the command is correct according to
the documentation.

I will have a look into the CLIP "sniffer" logs to see what I can find.

FredLeudon, 2017-03-06:
Thanks for feed-back.

yoh-there, 2017-03-06:
CLIP tool seems to send 19023b. I cannot test right now, but I will later
today. BTW, this is NOT what the documentation says.

FredLeudon, 2017-03-06:
I will test it tomorrow early too. 

There are (I think) same issues with others ECU. 

I had to change the code for reading dtc's
I believe that bcb ecu have 2 dtc command (semicolon separated) to read dtcs. 
So I loop with each command obtained by ecu.getgetdtcs()

yoh-there, 2017-03-06:
Correct, as the BCB is "2 in 1". The definition files came in two also: one
is called BCB, the other JB2 (Junction box). EVC is one though, as far as I
can tell.

FredLeudon, 2017-03-06:
I have a little request :

Could you explain dtc flags as they are given by CanZE ?
I would like to make them more understandable by users (and translate them too)

Thanks

yoh-there, 2017-03-06:
Of course. The flags are located in the fourth byte of the sequence. You
can find is coded in the Dtcs class
https://github.com/fesch/CanZE/blob/master/app/src/main/java/lu/fisch/canze/actors/Dtcs.java
and here is a good description
http://raviprashanthsn-technical.blogspot.nl/2013/10/fault-code-status-bits-in-automotive.html

if ((flags & 0x01) != 0) result += ", tstFail";
if ((flags & 0x02) != 0) result += ", tstFailThisOp";
if ((flags & 0x04) != 0) result += ", pendingDtc";
if ((flags & 0x08) != 0) result += ", confirmedDtc";
if ((flags & 0x10) != 0) result += ", noCplSinceClear";
if ((flags & 0x20) != 0) result += ", faildSinceClear";
if ((flags & 0x40) != 0) result += ", tstNtCpl";
if ((flags & 0x80) != 0) result += ", wrnLght";

*Bit0* => This Bit is “testFailed”. This bit provides the information about
the fault (Error) is still active (injected) or not.     If Fault is still
injected or Active, then the value is 1 otherwise the value is
0.              Ex: Fault “Short Circuit” is still active.

*Bit1* =>  This Bit is “testFailedThisOperationCycle”. This bit indicates
whether the fault is occurred anytime during the current operation cycle.
If Fault has occurred in the current operation cycle, then the value is 1
otherwise the value is 0.

*Bit2*  => This Bit is “pendingDTC”. This bit indicates whether the fault
is occurred anytime during the current operation cycle. The only difference
between “testFailedThisOperationCycle” and “pendingDTC” is
“testFailedThisOperationCycle” is cleared at the end of current operation
cycle (least bothering whether the fault is still active or inactive) and
“pendingDTC” is cleared only when in next operation cycle the monitor
routine is run and the result shows pass(fault is not present). So if Fault
is still injected or Active in the current operation cycle, then the value
is 1 otherwise if the Fault was active in previous operation cycle and is
inactive (i.e monitor routine is run and fault is inactive) in the current
operation cycle, then the value is 0.

*Bit3*  => This Bit is “confirmedDTC”. This bit informs that fault is
continuously active for specific monitor routines and is matured enough in
the current operation cycle so that it can be said Confirmed. If fault is
active and matured, then the value is 1 otherwise the value is 0.

*Bit4 * => This Bit is “testNotCompletedSinceLastClear”. This bit informs
that monitor routine is not run in the current operation cycle(once after
Clearing the DTC is done). The reason for not running in the current
operation cycle can be because particular pin is inactive in the operation
cycle(Ex: Parked or hibernate vehicle mode). If the monitor routine is not
completed this operation cycle, then the value is 1 otherwise the value is
0.

*Bit5*  => This Bit is “testFailedSinceLastClear”. This bit informs monitor
routine has reported that test has failed (at least once Bit0 is set) in
any operation cycle at least once after clearing the DTC action is
performed. If the fault has occurred after clear DTC is performed, then the
value is 1 otherwise the value is 0.

*Bit6*  => This Bit is “testNotCompletedThisOperationCycle”. This bit
informs that the monitor routine is still not run during this current
operation cycle. This is because the pin is not active for this operation
cycle or when the request is sent from the tester, the monitor routine is
not run. If the monitor routine is not run this operation cycle, then the
value is 1 otherwise the value is 0.

*Bit7*  => This Bit is “warningIndicatorRequested”. This bit is used to
bring into the attention of the user or driver when the fault occurs. If
fault occurs and any monitor is required for specific fault, then the value
is 1 otherwise the value is 0.

FredLeudon, 2017-03-06:
Thanks a lot !

FredLeudon, 2017-03-07:
hi, I tried 0319023b and still receiving NO DATA response.

07/03/2017	06:31:47,70	     	[]	cDtcActivity	doQueryECU(ecu[Electric Vehicle Controller-EVC-(renault ID:946)]
07/03/2017	06:31:47,75	     	[]	cDtcActivity	doQueryECU()	ecu.getGetDtcs() = 59023b
07/03/2017	06:31:47,80	     	[]	cDevice	requestFrame[7ec.59023b]
07/03/2017	06:31:47,81	     	[]	Device	requestIsoTpFrame : [7e4]
07/03/2017	06:31:47,83	     	[]	cELM327_WiFi	sendCommand(command[atsh7e4], timeout[350], addReturn[1], waitmillis[0], stopCar[>]
07/03/2017	06:31:47,95	     	[]	cELM327_WiFi	sendCommand() = [OK

>]	temps de traitement [122] ms
07/03/2017	06:31:47,95	     	[]	cELM327_WiFi	sendCommand(command[atfcsh7e4], timeout[350], addReturn[1], waitmillis[0], stopCar[>]
07/03/2017	06:31:48,07	     	[]	cELM327_WiFi	sendCommand() = [OK

>]	temps de traitement [116] ms
07/03/2017	06:31:48,08	     	[]	cELM327_WiFi	sendCommand(command[0319023b], timeout[750], addReturn[1], waitmillis[5], stopCar[>]
07/03/2017	06:31:48,30	     	[]	cELM327_WiFi	sendCommand() = [NO DATA

>]	temps de traitement [215] ms
07/03/2017	06:31:48,31	     	[]	cDevice	requestIsoTpFrame() -> QueryTime[00:00:501]
07/03/2017	06:31:48,33	     	[]	Device: request for 7ec.59023b returned error -E-data empty

yoh-there, 2017-03-07:
Yeah. Bummer. I will experiment a bit with what CLIP did, trying to emulate
it. Interesting how we moved from problematic BCB to EVC ;-)

yoh-there, 2017-03-07:
This will help. I will experiment with the "Extended diag" command

> 18.044016 CXX GVRET-PC Reverse Engineering Tool Output V164
> 18.044016 R11 000007E4 02 10 C0 00 00 00 00 00 Extended diag
> 18.051745 R11 000007EC 02 50 C0 AA AA AA AA AA 
> 18.056691 R11 000007E4 02 21 80 00 00 00 00 00 Part number
> 18.061271 R11 000007EC 10 1A 61 80 30 30 39 31 
> 18.063291 R11 000007E4 30 00 00 00 00 00 00 00 Flow control
> 18.064451 R11 000007EC 21 52 05 30 30 31 30 30 
> 18.070439 R11 000007EC 22 35 34 52 02 03 06 80 
> 18.074827 R11 000007EC 23 03 85 01 01 01 88 AA 
> 18.083006 R11 000007E4 03 22 33 A5 00 00 00 00 Compressor configuration
> 18.101366 R11 000007EC 04 62 33 A5 01 AA AA AA Denso ES27
> 18.114521 R11 000007E4 02 10 81 00 00 00 00 00 Diag session
> 18.121205 R11 000007EC 02 50 81 AA AA AA AA AA 
> 18.220862 R11 000007E4 02 10 C0 00 00 00 00 00 Extended diag
> 18.231711 R11 000007EC 02 50 C0 AA AA AA AA AA 
> 18.463957 R11 000007E4 03 19 02 3B 00 00 00 00 DTC's
> 18.711833 R11 000007EC 10 F3 59 02 FF 04 90 72 
> 18.713840 R11 000007E4 30 00 00 00 00 00 00 00 Flow control
> 18.715148 R11 000007EC 21 50 04 90 F1 50 04 11 041168 is one
> 18.720483 R11 000007EC 22 68 28 04 65 13 50 04 
> 18.724382 R11 000007EC 23 65 11 50 04 65 12 50 
> 18.729621 R11 000007EC 24 01 41 11 50 01 41 12 
> 18.734818 R11 000007EC 25 50 04 64 11 50 04 64 
> 18.740405 R11 000007EC 26 12 50 04 61 13 50 04 
> 18.744403 R11 000007EC 27 61 11 50 04 61 12 50 
> 18.749603 R11 000007EC 28 01 06 17 50 01 06 16 
> 18.754774 R11 000007EC 29 50 01 05 49 50 00 22 
> 18.760117 R11 000007EC 2A 7A 50 00 23 96 50 00 
> 18.764408 R11 000007EC 2B 23 13 50 00 23 F1 50 
> 18.770041 R11 000007EC 2C 00 23 F2 50 01 40 11 
> 18.774826 R11 000007EC 2D 50 01 40 12 50 06 0A 
> 18.780203 R11 000007EC 2E F1 50 06 0A F2 50 15 1525F3 is one
> 18.784493 R11 000007EC 2F 25 F3 2B 08 33 62 50 
> 18.791887 R11 000007EC 20 04 90 96 50 04 60 64 
> 18.794792 R11 000007EC 21 50 04 80 13 50 00 26 
> 18.800603 R11 000007EC 22 17 50 00 26 16 50 00 
> 18.804468 R11 000007EC 23 26 49 50 05 03 13 50 
> 18.810055 R11 000007EC 24 06 01 92 68 04 91 11 060192 is one
> 18.814829 R11 000007EC 25 50 04 92 11 50 03 05 
> 18.820533 R11 000007EC 26 94 50 00 22 08 50 00 
> 18.824408 R11 000007EC 27 26 81 50 00 22 7B 50 
> 18.829623 R11 000007EC 28 00 23 7A 50 00 23 11 
> 18.834829 R11 000007EC 29 50 00 23 12 50 00 20 
> 18.840124 R11 000007EC 2A 62 50 00 20 17 50 00 
> 18.844500 R11 000007EC 2B 20 16 50 00 27 17 50 
> 18.849637 R11 000007EC 2C 00 27 16 50 00 20 F3 
> 18.854822 R11 000007EC 2D 50 00 22 86 50 01 80 
> 18.860172 R11 000007EC 2E 92 50 05 00 67 50 05 0510F1 is one (but not shown in report)
> 18.864434 R11 000007EC 2F 10 F1 28 04 82 72 50 
> 18.869980 R11 000007EC 20 04 81 73 50 04 81 13 
> 18.874801 R11 000007EC 21 50 04 81 11 50 04 81 
> 18.880087 R11 000007EC 22 12 50 05 60 63 50 AA 
> 18.970234 R11 000007E4 03 22 20 06 00 00 00 00 Odo
> 18.991658 R11 000007EC 06 62 20 06 00 E6 C0 AA 
> 19.012634 R11 000007E4 03 22 32 06 00 00 00 00 SOH
> 19.031654 R11 000007EC 04 62 32 06 54 AA AA AA 
> 19.058825 R11 000007E4 03 22 33 DC 00 00 00 00 Total energy consumed
> 19.071949 R11 000007EC 06 62 33 DC 90 4F 08 AA 
> 19.082844 R11 000007E4 02 10 81 00 00 00 00 00 Diag session
> 19.091869 R11 000007EC 02 50 81 AA AA AA AA AA 
> 28.445394 R11 000007E4 02 10 C0 00 00 00 00 00 Extended diag
> 28.451428 R11 000007EC 02 50 C0 AA AA AA AA AA 
> 28.459095 R11 000007E4 02 21 80 00 00 00 00 00 Part number
> 28.471641 R11 000007EC 10 1A 61 80 30 30 39 31 
> 28.473788 R11 000007E4 30 00 00 00 00 00 00 00 Flow control
> 28.474580 R11 000007EC 21 52 05 30 30 31 30 30 
> 28.479154 R11 000007EC 22 35 34 52 02 03 06 80 
> 28.484203 R11 000007EC 23 03 85 01 01 01 88 AA 
[... 56 lines cut ...]
> 29.840948 R11 000007EC 10 08 59 06 04 11 68 28 
> 29.843117 R11 000007E4 30 00 00 00 00 00 00 00 Flow control
> 29.851407 R11 000007EC 21 84 04 AA AA AA AA AA 
> 29.862054 R11 000007E4 06 19 06 04 11 68 80 00 3rd context 041168
> 29.891392 R11 000007EC 10 0A 59 06 04 11 68 28 
> 29.893505 R11 000007E4 30 00 00 00 00 00 00 00 Flow control
> 29.901054 R11 000007EC 21 80 00 E6 03 AA AA AA 
> 29.957893 R11 000007E4 06 19 06 05 10 F1 84 00 2nd context 0510F1
> 29.991399 R11 000007EC 10 08 59 06 05 10 F1 28 
> 29.993871 R11 000007E4 30 00 00 00 00 00 00 00 Flow control
> 30.000952 R11 000007EC 21 84 01 AA AA AA AA AA 
> 30.004935 R11 000007E4 06 19 06 05 10 F1 80 00 3rd context 0510f1
> 30.040917 R11 000007EC 10 0A 59 06 05 10 F1 28 
> 30.042875 R11 000007E4 30 00 00 00 00 00 00 00 Flow control
> 30.051330 R11 000007EC 21 80 00 E6 03 AA AA AA 
> 30.170138 R11 000007E4 02 21 81 00 00 00 00 00 VIN
> (removed)
> 30.199144 R11 000007E4 02 21 80 00 00 00 00 00 Partnumber
> 30.211365 R11 000007EC 10 1A 61 80 30 30 39 31 
> 30.213315 R11 000007E4 30 00 00 00 00 00 00 00 Flow control
> 30.214459 R11 000007EC 21 52 05 30 30 31 30 30 
> 30.220447 R11 000007EC 22 35 34 52 02 03 06 80 
> 30.224074 R11 000007EC 23 03 85 01 01 01 88 AA 
> 30.252911 R11 000007E4 02 10 81 00 00 00 00 00 Default diag
> 30.260964 R11 000007EC 02 50 81 AA AA AA AA AA

FredLeudon, 2017-03-07:
If some ECU needs extended diag command, perhaps will it be a good idea to add then on ECU/ECUS class ?

yoh-there, 2017-03-07:
Absolutely!!!

The EVC is still resisting me though ;-) But I will break it, Moo haa haa!!

FredLeudon, 2017-03-08:
I'm trying this morning ...

08/03/2017	10:33:27,75	     	[]	Test : EVC DTC reading
08/03/2017	10:33:27,75	     	[]	cELM327_WiFi	sendCommand(command[atsh7e4], timeout[500], addReturn[1], waitmillis[0], stopCar[]
08/03/2017	10:33:28,28	     	[]	cELM327_WiFi	sendCommand() = [OK]	temps de traitement [530] ms
08/03/2017	10:33:28,28	     	[]	cELM327_WiFi	sendCommand(command[atfcsh7e4], timeout[500], addReturn[1], waitmillis[0], stopCar[]
08/03/2017	10:33:28,79	     	[]	cELM327_WiFi	sendCommand() = [OK]	temps de traitement [515] ms
08/03/2017	10:33:28,80	     	[]	cELM327_WiFi	sendCommand(command[0210c0], timeout[500], addReturn[1], waitmillis[0], stopCar[]
08/03/2017	10:33:29,31	     	[]	cELM327_WiFi	sendCommand() = [0250C0AAAAAAAAAA]	temps de traitement [515] ms
08/03/2017	10:33:29,31	     	[]	cELM327_WiFi	sendCommand(command[atsh7e4], timeout[500], addReturn[1], waitmillis[0], stopCar[]
08/03/2017	10:33:29,83	     	[]	cELM327_WiFi	sendCommand() = [OK]	temps de traitement [516] ms
08/03/2017	10:33:29,83	     	[]	cELM327_WiFi	sendCommand(command[atfcsh7e4], timeout[500], addReturn[1], waitmillis[0], stopCar[]
08/03/2017	10:33:30,34	     	[]	cELM327_WiFi	sendCommand() = [OK]	temps de traitement [515] ms
08/03/2017	10:33:30,34	     	[]	cELM327_WiFi	sendCommand(command[022180], timeout[500], addReturn[1], waitmillis[0], stopCar[]
08/03/2017	10:33:30,87	     	[]	cELM327_WiFi	sendCommand() = [101A6180303039312152053030313030223534520203068023038501010188AA]	temps de traitement [530] ms
08/03/2017	10:33:30,87	     	[]	cELM327_WiFi	sendCommand(command[atsh7e4], timeout[500], addReturn[1], waitmillis[0], stopCar[]
08/03/2017	10:33:31,41	     	[]	cELM327_WiFi	sendCommand() = [OK]	temps de traitement [536] ms
08/03/2017	10:33:31,41	     	[]	cELM327_WiFi	sendCommand(command[atfcsh7e4], timeout[500], addReturn[1], waitmillis[0], stopCar[]
08/03/2017	10:33:31,94	     	[]	cELM327_WiFi	sendCommand() = [OK]	temps de traitement [530] ms
08/03/2017	10:33:31,94	     	[]	cELM327_WiFi	sendCommand(command[032233a5], timeout[500], addReturn[1], waitmillis[0], stopCar[]
08/03/2017	10:33:32,47	     	[]	cELM327_WiFi	sendCommand() = [046233A501AAAAAA]	temps de traitement [530] ms
08/03/2017	10:33:32,47	     	[]	cELM327_WiFi	sendCommand(command[atsh7e4], timeout[500], addReturn[1], waitmillis[0], stopCar[]
08/03/2017	10:33:32,98	     	[]	cELM327_WiFi	sendCommand() = [OK]	temps de traitement [516] ms
08/03/2017	10:33:32,99	     	[]	cELM327_WiFi	sendCommand(command[atfcsh7e4], timeout[500], addReturn[1], waitmillis[0], stopCar[]
08/03/2017	10:33:33,51	     	[]	cELM327_WiFi	sendCommand() = [OK]	temps de traitement [515] ms
08/03/2017	10:33:33,51	     	[]	cELM327_WiFi	sendCommand(command[021081], timeout[500], addReturn[1], waitmillis[0], stopCar[]
08/03/2017	10:33:34,02	     	[]	cELM327_WiFi	sendCommand() = [025081AAAAAAAAAA]	temps de traitement [515] ms
08/03/2017	10:33:34,02	     	[]	cELM327_WiFi	sendCommand(command[atsh7e4], timeout[500], addReturn[1], waitmillis[0], stopCar[]
08/03/2017	10:33:34,55	     	[]	cELM327_WiFi	sendCommand() = [OK]	temps de traitement [531] ms
08/03/2017	10:33:34,55	     	[]	cELM327_WiFi	sendCommand(command[atfcsh7e4], timeout[500], addReturn[1], waitmillis[0], stopCar[]
08/03/2017	10:33:35,08	     	[]	cELM327_WiFi	sendCommand() = [OK]	temps de traitement [529] ms
08/03/2017	10:33:35,09	     	[]	cELM327_WiFi	sendCommand(command[0210c0], timeout[500], addReturn[1], waitmillis[0], stopCar[]
08/03/2017	10:33:35,60	     	[]	cELM327_WiFi	sendCommand() = [0250C0AAAAAAAAAA]	temps de traitement [514] ms
08/03/2017	10:33:35,60	     	[]	cELM327_WiFi	sendCommand(command[atsh7e4], timeout[500], addReturn[1], waitmillis[0], stopCar[]
08/03/2017	10:33:36,13	     	[]	cELM327_WiFi	sendCommand() = [OK]	temps de traitement [531] ms
08/03/2017	10:33:36,13	     	[]	cELM327_WiFi	sendCommand(command[atfcsh7e4], timeout[500], addReturn[1], waitmillis[0], stopCar[]
08/03/2017	10:33:36,65	     	[]	cELM327_WiFi	sendCommand() = [OK]	temps de traitement [515] ms
08/03/2017	10:33:36,65	     	[]	cELM327_WiFi	sendCommand(command[0319023b], timeout[500], addReturn[1], waitmillis[0], stopCar[]
08/03/2017	10:33:37,18	     	[]	cELM327_WiFi	sendCommand() = [NO DATA]	temps de traitement [531] ms

yoh-there, 2017-03-08:
Same results here. Very frustrating!

FredLeudon, 2017-03-25:
Hi,

If possible, could You send me a full (from beginning to complete) Clip diagnostic reverse log by mail, I would like to reproduce EVC query by clip.
Perhaps is there a command sequence we didn't see. 
My mail is ioscanze[at]icloud[dot]com

Thanks

yoh-there, 2017-03-25:
There you go Fred

FredLeudon, 2017-03-28:
I would like to reproduce such document.

[DiagZoéClip.pdf](https://github.com/fesch/CanZE/files/874779/DiagZoeClip.pdf)

yoh-there, 2019-01-13:
I am closing this issue due to lack of activity.

closed 2019-01-13 by yoh-there