fesch/CanZE #407: V 1.20 crashes

MAlfare, 2017-01-29:
Hi!
Crashes after displaying "connected to..." and on KONNWEI KW902 BT is on, Host flashes, OBD off.
Lenovo A3500 F Tablet, Android 4.4.2 and Motorola Moto G4+, Android 6.0.1.
Both worked with previous version.

fesch, 2017-01-29:
What locales do you have in your settings?

yoh-there, 2017-01-29:
Damn, I thought there was always fallback to the default language? Is that just in the XML layouts??

MAlfare, 2017-01-29:
@fesch: The language on my Android devices?
It is "Deutsch"

fesch, 2017-01-29:
I've put my Android on "Deutsch" --> CanZE remains EN
If I chose "Français" --> CanZE becomes FR :-)

So no crash on my device!

yoh-there, 2017-01-29:
Same here. Confused?

fesch, 2017-01-29:
@MAlfare : What options do you have selected in the settings tab?
What happens if you switch you Android language to English?

I need to be able to reproduce the crash ...

yoh-there, 2017-01-29:
Seems to fail only if SD card logging -> Fields is on, independant of Locale. That is my first analysis at least.

fesch, 2017-01-29:
Indeed. That is reproducible. 

I've uploaded a fix. Can you confirm it works at your side too?

yoh-there, 2017-01-29:
Pulling. Hold on........

yoh-there, 2017-01-29:
Fix confirmed!

MAlfare, 2017-01-29:
My settings, language set to English(UK)
[image]
[image]

fesch, 2017-01-29:
Version 1.21 with fix has been uploaded and is waiting for validation and dispatching in the store.

closed 2017-01-29 by fesch