fesch/CanZE #410: Use field "Range" in "Battery" screen for actual SoC %

GertjanVB, 2017-01-29:
Since "Range" is displayed quite  clearly  in the dash of ZOE, I suggest to use this space in the "Battery" screen for the exact SoC %. SoC is displayed in the graph on the "Battery" screen, but the exact figure at the moment is not easy to determine while driving. Displaying the exact SoC % in this data field will give a better understanding off the graphic and better impression on remaining range and duration of charging. I use CanZE for iOS.

yoh-there, 2017-01-29:
I think you're referring to the iOS version right? As the Android version shows State of Health in numbers, State of Charge (Real and Usable) in a timed graph, Cell voltages as a line graph and  Battery temperatures ditto. Range is not stated. I am asking to ensure proper assignment and possibly ensuring what is shown is equal in both versions.

If possible, add a screen shot please. Thanks

GertjanVB, 2017-01-30:
Yep iOS version:
[image]

FredLeudon, 2017-01-30:
Do You prefer Soc or User Soc ?

GertjanVB, 2017-01-30:
SoC, in connection to the SoC displayed in the dash, when charging is started. 
Hmm, what is user SoC exactly? SoH * SoC or % of the total real battery capacity, or something completely different?

FredLeudon, 2017-01-30:
@CanZE Team, could You explain the difference between UserSoc and SoC please ?

HarmOtten, 2017-01-31:
See my comment about SoC versus User SoC: [https://github.com/fesch/CanZE/issues/412#issuecomment-276454236](url)

FredLeudon, 2017-01-31:
I will display User SOC in battery activity

closed 2017-01-31 by FredLeudon