fesch/CanZE #412: SOC vs. User SOC

fesch, 2017-01-30:
This is no bug, just a question:

what is the difference between the field "SOC" and the field "User SOC".

yoh-there, 2017-01-30:
I don't know for sure, this is simply how the fields are called in the documentation files. @HarmOtten has played with them a bit and I think his conclusion was that real was the "raw battery" SOC. Let me drop him a mail and ask him.

yoh-there, 2017-01-30:
It seems both percentages do not change with SOH.

A "bad" battery will charge to 100% User SOC, in other words, it is the percentage of it's usable capacity, which is the raw capacity minus the head room and "foot room". Unfortunately we don't know how these limits are sized when the SOH goes down, but it is fair to assume they will go down.

Real SOC seems to be raw state of charge of the battery, as percentage of it's total, current capacity. The latter probably being 25.92 * SOH% kWh.

This is not yet substantiated by hard numbers, but our working hypothesis.

HarmOtten, 2017-01-31:
I think what fesch means, are the two lines on the battery-screen. Both are the usable SOC %. The difference is that the [User SOC] has 2 decimals places and therefore more precise, and the [SOC] is the rounded value displayed on the dash.
Maybe it is a nice idea to add the [Real SOC %] to this graph as a thin grey line.

yoh-there, 2017-02-02:
There are a lot of SOC fields in the car. I am not sure which ones you are referring to but

42e.0 => resolution is 0.02. Source is EVC, to BCM and TCU
654.25 => resolution is 1, it's actually called "Energy level", sourceis EVC, to dash and TCU

So yeah, I think Harm is right that 654 is for the dashboard, but I am not 100% sure 42e is simply a more precise value.

Close? ;-)

yoh-there, 2017-02-08:
I compared both and indeed the 654.25 is a rounded value of 42e.0. Closing.

closed 2017-02-08 by yoh-there