fesch/CanZE #414: Menu CHARGING (Android V1.21)

STEN-ZOE-FAN, 2017-01-31:
In subsection "Battery modules temperatures and balancing" are actually only the module Temperatures shown. Balancing info is allways "0".

yoh-there, 2017-01-31:
Not for my car, at least not always :-) According to the documentation we have this is indeed balancing. It might be your battery is nicely balanced, or you have a different software version. Could you please go to the firmware menu, tap the LBC and attach a screen print here? I can try to figure that out with the 4 version information strings at the bottom of the screen. Also please indicate the model of the car (Q210, R240, Q90, R90).

yoh-there, 2017-02-08:
Closing due to lack of feedback. Feel free to reopen.

closed 2017-02-08 by yoh-there