fesch/CanZE #415: Menu FIRMWARE (Android V1.21)

STEN-ZOE-FAN, 2017-01-31:
No version-infos are shown, only the items EVC / TCU ... are displayed.

yoh-there, 2017-01-31:
Can you please add a screen print, indicate your car type and if you're using iOS or Android? Thank you!

Albino-Zoe, 2017-02-02:
I had the same problem ;-) The doing in V1.21 is now different. Select the item EVC / TCU, then the value appears.

yoh-there, 2017-02-02:
Thanks for confirming. I will test later. Some ECU's are a bit stubborn, and the BCB never gives up any data, but only the EVC or TCU, that is not right. Can you add please the type of car (Q210, R240, Q90, R90)?

yoh-there, 2017-02-02:
I cannot reproduce this, sorry. Please ensure that the car is unlocked, card inserted (if not a keyless version) and "Started", as this awakens all ECU's.

When I do that I can query the firmware for nearly all ECU's.

I would appreciate if you can repeat the procedure with the above in mind, and report the results.

Please note that the BCB almost never answers, this is not bug.

Thanks.

yoh-there, 2017-02-08:
Closing due to lack of feedback. Feel free to reopen.

closed 2017-02-08 by yoh-there