fesch/CanZE #416: Menu RANGE (Android V1.21)

STEN-ZOE-FAN, 2017-01-31:
Remaining range by car is correctly shown, but the following fields
(Remaining Range by CanZE) are displaying mystery values.

yoh-there, 2017-01-31:
Tap that line and enter your distance to destination. Please indicate if this issue can be closed.

yoh-there, 2017-02-08:
I have misunderstood you I now realize. @fesch, this is about Technical -> Range. I am assigning this to you. @STEN-ZOE-FAN has a point ;-)

fesch, 2017-02-08:
I know. This screen is not yet finished. I did not have the time yet, but I will finished it soon ;)

FredLeudon, 2017-02-08:
[image]

Strange, when battery soc approch 0, max consumpsion change to 0.5kWh/100 km

yoh-there, 2017-02-08:
Interesting observation. Maybe that is that tiny bit extra you get into what I call the "foot-room" (as opposed to "head-room"), so into the non-usable part of the battery, to get to a nearby charger in low performance mode? Pure speculation I know :-) Still, half a kWh on low speed should give you at least 8km.........

solmoller, 2017-02-09:
Interesting. Did the car actually report limited mode?

<https://github.com/fesch/CanZE/issues/416#issuecomment-278370266>, or mute
the thread

FredLeudon, 2017-02-09:
Yes, limited mode.

Arriving with 0 kWh in batt ;-)

I forgot to take a picture at arrival.

Frédéric RICHARD

solmoller, 2017-02-09:
Ok.  My experience is that SOH suffers from that treatment

BR
Henrik

zmeglic, 2017-02-14:
What would this remaining range by CanZE supposed to be displaying? These large numbers make no sense.
[image]

fesch, 2017-02-14:
I will delete the entire screen.
Too many question about a screen that is technical and not yet finished.
Sorry.

closed 2017-02-14 by fesch