fesch/CanZE #418: Add Slovenian

zmeglic, 2017-01-31:
Here is my contribution. Translation to Slovenian language. (Slovenščina)
There might be a few typoos, but I will fix them if I find them. 
Also these three are not translated, because I have no idea what they mean.
<item>Blow req</item>
        <item>Cool cond req</item>
        <item>Heat cond req</item>
 
Hope you can include it soon.
Regards, Zdenko

[SL_strings.xml.zip](https://github.com/fesch/CanZE/files/742205/SL_strings.xml.zip)

yoh-there, 2017-01-31:
That is very kind, thank you, we will surely include it in the next version!

- Blower requested
- Cooling requested
- Heating requested

But it's OK to leave it like that.

yoh-there, 2017-01-31:
Added in the translation branch. Let's not forget to merge before the next release!

closed 2017-01-31 by yoh-there

zmeglic, 2017-01-31:
OK, than I will translate these together with any corrections/future translations at the next update.

zmeglic, 2017-02-23:
How do I update the translations when new pages are added?

yoh-there, 2017-02-23:
The easiest way is to pick up the file here 

https://github.com/fesch/CanZE/blob/master/app/src/main/res/values-sl/strings.xml

If I am not mistaken we have put in the new strings as English and if you could be so kind to simply change the few lines that are, we can easily merge that in. Thank you!

zmeglic, 2017-02-27:
New file, merged with english variant:

[strings.zip](https://github.com/fesch/CanZE/files/803334/strings.zip)

yoh-there, 2017-02-28:
Thank you! Will merge it in.