fesch/CanZE #419: Range incorrect on Battery-screen

HarmOtten, 2017-01-31:
(iOS / ZOE R90)
At low SOC (below 4%) the Range suddenly spikes to 1023km. At higher SOC it was displayed correctly.
[image]

fesch, 2017-01-31:
As far as I remember, "1023" is a special value indicating that no calculation was possible. The car would indicate "---" and so should do CanZE as well.

FredLeudon, 2017-01-31:
Thanks for feedback.

In next release, Range will be replaced by User SOC.

All Range calculation have been checked.
Will display --- in text or will be assigned to -1 in graphics when value >= 1023

closed 2017-01-31 by FredLeudon

yoh-there, 2017-01-31:
Correct. Almost all values supplied by the ECUs are given as unsigned integers (sometimes offset such as with temperatures) and most often the highest possible value formally means "not available". Very few numbers are represented as signed integers. This is indicated by the SIGNED bit in the options attribute of a FIELD.